gioblu/PJON #349: bus.reply fail to send before ESP.restart();

fotosettore, 2020-04-30:
hi Giovanni
i need to restart a slave when requested from master.
and before restart, i send to master tha message that slave is resetting
i use this code
```
...
bus.reply("RESET"), 5);
delay (2000);
ESP.restart();
```
but it does not work
I tried to comment ESP.restart() and the message is send.
I tried to use a "millis" routine but it does not work too.

Any idea to solve?

ciao

gioblu, 2020-05-01:
Ciao @fotosettore I think the problem is you don't call `update` after `reply`, so the packet is dispatched in the buffer but never transmitted.

fotosettore, 2020-05-01:
i'm sorry
it does not work :-(
`...
bus.reply("RESET"), 5);
bus.update();
delay (2000);
ESP.restart();`

any more idea ?

gioblu, 2020-05-01:
Ciao @fotosettore `update` returns the number of packets that are still to be transmitted, when its return value is 0 means all packets dispatched were transmitted. Instead of calling a `delay` try to call update until it returns 0, it should fine then.

fotosettore, 2020-05-01:
Well done Giovanni !
Many thanks for your input !
Now, using this code, all works fine
```
          unsigned long LastResetTime = millis();
          while ( bus.update() != 0 )
          {
              if (millis() - LastResetTime > 2000)
              {
                 break;
              }
          }
         
          ESP.restart();
```
I also used **millis** to force exit if any problem about communication happen.
Ciao
Peppe

gioblu, 2020-05-02:
Ciao @fotosettore thank you, consider that when the maximum amount of retries is reached the packet is discarded, so `update` will return 0, no risk of looping in there forever :)

closed 2020-05-02 by gioblu

fotosettore, 2020-05-02:
so ... how many maximum amount of retries are allowed ?

gioblu, 2020-05-03:
Ciao @fotosettore if using SoftwareBitBang it takes around 3.5 seconds and 20 retries.