gioblu/PJON #350: LocalUDP Receive causes intermittent abort on ESP32

justoke, 2020-05-06:
I have been chasing down ESP32 reboots and this is something I observe intermittently on the LocalUDP strategy. Obviously reboots of this nature are not desirable, so I'm wondering if any defensive strategy can be taken to safeguard the strategy.

Abort message:

> abort() was called at PC 0x40163e5f on core 1
> Backtrace: 0x4008c8b0:0x3ffb1d30 0x4008cae1:0x3ffb1d50 0x40163e5f:0x3ffb1d70 0x40163ea6:0x3ffb1d90 0x4014b087:0x3ffb1db0 0x4014b176:0x3ffb1dd0 0x4014b12d:0x3ffb1df0 0x40165f0f:0x3ffb1e10 0x400e278e:0x3ffb1e50 0x400e284b:0x3ffb1e90 0x400e2c2d:0x3ffb1ec0 0x400e2f7f:0x3ffb1f30 0x400e5d27:0x3ffb1f50 0x400d330a:0x3ffb1f80 0x400e8fd5:0x3ffb1fb0 0x40088ff9:0x3ffb1fd0

Which can be decoded as follows: 
```
Decoding stack results
0x4008c8b0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008cae1: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x40163e5f: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x40163ea6: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4014b087: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x4014b176: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x4014b12d: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
0x40165f0f: WiFiUDP::parsePacket() at C:\Users\sbens\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiUdp.cpp line 210
0x400e278e: UDPHelper::receive_frame(unsigned char*, unsigned short) at .pio\libdeps\esp32dev\PJON_ID245\src/strategies/LocalUDP/../../interfaces/ARDUINO/UDPHelper_ARDUINO.h line 41
0x400e284b: DualUDP::receive_frame(unsigned char*, unsigned short) at .pio\libdeps\esp32dev\PJON_ID245\src/strategies/DualUDP/DualUDP.h line 254
0x400e2c2d: PJON ::receive() at .pio\libdeps\esp32dev\PJON_ID245\src/PJON.h line 275
0x400e2f7f: service_messages() at lib\messaging\messaging.cpp line 157
0x400e5d27: StationManager::DoWork() at lib\StationManager\StationManager.cpp line 547
0x400d330a: loop() at src\main.cpp line 81
0x400e8fd5: loopTask(void*) at C:\Users\sbens\.platformio\packages\framework-arduinoespressif32\cores\esp32\main.cpp line 19
0x40088ff9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

gioblu, 2020-05-06:
Ciao @justoke thank you for your report, I will look into it and alert @fredilarsen

fredilarsen, 2020-05-06:
On line 210 of the current WiFiUDP.cpp for ESP32 on github, it does a `new char[1460]`.

@justoke Can you check if this is the same code as in your copy of WiFiUDP.cpp?

It seems the C++ library gets panic when trying to allocate the memory. It does not explicitly say it is an out-of-memory situation but that would be the most probable cause. Are you in a tight memory situation? Do you think there could be a memory leak located in your code, or does it have to be in a C++/system library or PJON?

fredilarsen, 2020-05-06:
Regarding the 3 UDP related strategies in PJON, DualUDP and GlobalUDP were created first to cover LAN and WAN cases, each with its strengths and weaknesses (GlobalUDP needing to know the IP and id of every device to talk to, LocalUDP stressing wireless networks by broadcasting everything).

Therefore the DualUDP was created to combine the autodiscovery-part of LocalUDP and the non-broadcasting and WAN part of GlobalUDP. It starts as LocalUDP (broadcasts) but changes to use directed packets when devices are located. Also allows devices on other network segments to be manually added.

I guess we could really deprecate LocalUDP and GlobalUDP, as my recommendation is to use DualUDP.

justoke, 2020-05-06:
> On line 210 of the current WiFiUDP.cpp for ESP32 on github, it does a `new char[1460]`.
> 
> @justoke Can you check if this is the same code as in your copy of WiFiUDP.cpp?
> 
> It seems the C++ library gets panic when trying to allocate the memory. It does not explicitly say it is an out-of-memory situation but that would be the most probable cause. Are you in a tight memory situation? Do you think there could be a memory leak located in your code, or does it have to be in a C++/system library or PJON?
> 
> As a side note, I am running a master using DualUDP on a Windows server, and it runs for months without problems with many packets every second.

Thank you for the prompt response. I'll check the code you are referring to.
This is the level of reliability I would like to achieve! Currently my app is far from running out of memory, but I will endeavour to track down anything which might be causing this behaviour. But I accept that my code is far from perfect and is at least to some degree less than optimal. I should mention I switched to DualUDP - I lost track of which one I was on.

justoke, 2020-05-07:
Yes, indeed there was a memory leak. I had not cleaned up a pointer in the message processing part of my code. I'll leave the stations running now and see how they perform. 

Thank you for your prompt response.

closed 2020-05-07 by justoke

justoke, 2020-05-07:
> Regarding the 3 UDP related strategies in PJON, DualUDP and GlobalUDP were created first to cover LAN and WAN cases, each with its strengths and weaknesses (GlobalUDP needing to know the IP and id [...]

I ended up using this by default. Having only one UDP strategy and one that covers most use cases would keep things simple.

fredilarsen, 2020-05-08:
@justoke Thank you for the feedback. It is always nice to know if advice was good or bad :-)
I hope your application will be running stable now that the memory problem is solved.

The other two strategies may have their use cases. LocalUDP for example requires much less memory as it has no tables and very little code, We are discussing whether to just flag DualUDP as the recommended of the 3, or to move the others to a "deprecated" or "special_use" subfolder or similar. Anyhow I think we should migrate all LocalUDP and GlobalUDP examples to DualUDP where not already present.