gioblu/PJON #351: Question: How to debug

andistancu, 2020-05-12:
Hello,

First of all, thank you for the amazing contribution to the society.
Second of all, I understand this is not a support forum, and I apologise for asking this question here. 

I have set up a nice SoftwareBitBang network and it works amazingly. As soon as I start using DigiSpark boards nothing happens. Can you please advise where to start debugging?
I reduced the complexity to the basic Local SoftwareBitBang example with one Receiver (Mega) and one Transmitter (DigiSpark). When I use the DigiSpark mega either doesn't receive anything, either I don't understand how to check. If I replace the DigiSpark with another Mega, it works just fine.

Thank you,
Andi Stancu

gioblu, 2020-05-12:
Ciao @andistancu thank you for your compliments :)
Which pin you are using on digispark? Have you tried other pins?

The digispark board does not have an internal clock that is generally required for the operation.
Many users reported digispark working nominally with SoftwareBitBang, although the clock is not precise, so some boards may work, others not, consider that depending on how is powered the digispark board chooses between 16 (when powered externally) and 16.5Mhz (when powered via usb), so that may be the issue: http://digistump.com/wiki/digispark/tricks

>When the digispark boots up, it uses the USB signal to calibrate its clock to 16.5mhz. When powered by an external power supply this doesn't happen and the clock speed is closer to 16.0mhz, with roughly 10% accuracy.

I would suggest to use ATtiny85 with an external 16MHz clock to reach the best performance.

Consider that master is now not stable, stick with the latest release for now.

gioblu, 2020-05-12:
To debug I generally use an oscilloscope, if you don't have can you can just light up a LED when a packet is received or when a transmission failed, you can even drive a LED with the bus, so you can see flashes when packets are transmitted (as it is done for tx and rx serial pins on the arduino).

gioblu, 2020-05-12:
See also: https://github.com/gioblu/PJON/wiki/ATtiny-interfacing

andistancu, 2020-05-13:
Hi there @gioblu,

I'm so impressed that with all the time you need to put in this library you had time to answer my question as well. As soon as I'm typing this I'm going straight to your shop to support you.

We are powering it externally.
We tried with pin 1, and with pin 2 when we noticed is not working.
We tried to different boards, hoping the first one is fried.

We also attempted to try adding an external oscillator to the DigiSpark but I can't find any resources about how to do that. I asked already in a few forums. 

The problem with trying an ATTiny85 without the DigiSpark is that it would become a problem in the bigger picture. We want about 400 sensors, so preparing the chip, with oscillators, power regulator, etc 400 times seems like a lot of time and room for error.

I was testing the setup with the most basic example, to eliminate the possibility of code error. The example blinks the master when a specific message is received ("B"). The blink doesn't happen and I was curious how to better understand why. Is the message not sent, not received, or is the message different than what I'm expecting it to be.

As a backup plan, we consider using Mini Pro (the version with an onboard external oscillator) but that will have a significant space impact.

Once again,
Thank you!

gioblu, 2020-05-15:
Ciao @andistancu is USB absolutely required or you could be ok to program devices using iSP?

andistancu, 2020-05-15:
Hi @gioblu. Last night we switched to Pro Mini and now SoftwareBitBang works like a charm. Tonight I'll try AnalogSampling. I'm very excited. 

Thus, personally I am not interested in making the digiSpark work anymore, however I am willing to continue debugging if it would help you identify a possible defect/improvement, or to write a tutorial for the community.

Would you like me to try ISP programming?

Thanks again, I can't wait to go through the book :)

gioblu, 2020-05-15:
Ciao @andistancu thank you very much for your support to the project, I hope you will enjoy the book. I am happy you have fixed the issue already using the Arduino Pro Mini, that is for sure much more reliable.  AnalogSampling is really cool, I agree, I also need soon to make another iteration on it to simplify PJDLS. I Hope to see you in our gitter chat :) 

> Would you like me to try ISP programming?

Forget about it, I was considering which board to suggest you, but you have already found the solution.

If you have time to give a try to SoftwareBitBang it would be cool to know if you have nominal performance using also faster modes, see: https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang#performance

Happy tinkering :)

closed 2020-05-15 by gioblu