gioblu/PJON #355: Getting started with Master / Multiple slave over RS485 (ThroughSerial strategy)

HEvNeck, 2020-06-17:
Hi Gioblu,

I've been looking for weeks for a good and reliable communication-library before stumbling at PJON.

It seems just the library i need for connecting several slaves (4 at present) which do several tasks in my home (pump-control / data-aquisition / valve-control etc.) to one master-device.
- The slaves have to report acquired data and or status-messages to the master.
- Also the slaves have to execute some commands issued by the master.
- All at the initiative of the master (round-robin or timed or a mixture of those).
- All of the device are interconnected over a CAT5 cable of approximately 30m total length with use of max485 module (some of which have auto-tx and some with DE/RE pins).
- At serial level this network functions correctly for all slaves at present time; they communicate one-at a time perfectly to my PC. (only when i issue a command from the PC, the situation gets really messed up! ;-) That's why i was looking for a mere reliable protocol....
- All slaves are Arduino (Uno / Pro-Micro), the master will be an Arduino for now. 
(I was planning to use a Raspberry-pi as master this in future, but i don't like switching from Python to C++ and vise-versa all the time, so for the time being i'll stick to an Arduino for that.)

There are lots of communication libraries swarming around, but most of them are incomplete, not suited to my needs, to complex for my needs (eg. modbus), instable, barely documented or not maintained for years.

Anyhow, my problem is: "how to start?"; I cannot find good "hands-on" examples from where to further work out my solution. There are lots of references (of which some of them contain broken links to github (renamed or removed?)) when i search Google on "pjon", "master/slave", "RS485", but they are all bits & pieces. Also it is hard searching the right example on Git-hub as there is no brief description of what the example is supposed to do. 
At the moment i have to open from all examples the "sender"and "receiver" code and read the code to find out what their interaction actually is, and then conclude that it is supposed to be used with an other Strategy for example.

This morning i ordered your pjon-book actually. I hope to get a better overview from that and that there are hand-on examples in it. But i think I'll have to wait for another 2 weeks before i get it delivered at home and in the meanwhile i'm desperately eager to make a start.

By the way, this is (mostly) not a complaint!; So, just a tip to make your library more successfull and more widely spread and used!
- A product is "as good" as it documentation; At the moment, for a  lot of methods, you have to dig through the classes to find out what a method is for and how to use it. (eg. bus.reply() )
- Create a sort of selection roadmap for selecting the right strategy for the different physical media types / configuration types.
- Please document the examples a bit more with in-code comments and a readme.md telling what the example is supposed to do and in which case it can be used.

Anyway, for what i've seen for now, i'm very pleased with your concept and i like the architecture and coding-style!; So keep up the good job!!! 

Maybe you can give me something to start with in the meanwhile? I think the postings can be very helpful to other users as wel. If possible, i maybe can post some of my code... But it would be very specific i'm afraid (there would be al lot of hardware related stuff in it...)

Sorry for this very lengthy post ;-(
Thanks for your reaction already! and again; Keep up the good work!

Cheers!, Hans

gioblu, 2020-06-17:
Ciao @HEvNeck thank you for your support to the project and for your good suggestions.
We have really few examples using RS485, and they are not well documented as you underlined, although you can use them to make the first steps:

https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/ThroughSerial/RS485-Blink
https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/ThroughSerial/RS485-BlinkWithResponse

I will work to add the `README.md` files in the examples as you well suggested and will add missing parts of the documentation such as `reply`.

I suggest you to give a try to `SoftwareBitBang` because many users have determined that it is so reliable that in most cases, if there is no strong source of interference, it can substitute RS485.

I hope you will enjoy the publication, thank you again.
If you spot anything else please report it, I will fix it as soon as possible :)

HEvNeck, 2020-06-17:
Hi Gioblu,

Thanks for your very prompt response!
It is nice to notice that you take my comments seriously.

For what concerning SoftwareBitBang vs. ThroughSerial...; I do not completely see the relationship between those two. Moreover as you suggest to look at the examples TroughSerial/RS485-Blink(withResponse).

Could you please give some background information on this part?

When i make progress during development, and there are some new insights or possibilities, I will be glad to share them with you!

Thanks in advance again!
Hans

gioblu, 2020-06-17:
>Could you please give some background information on this part?

`SoftwareBitBang` is a strategy that implements the [PJDL](https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang) protocol, which I have invented and iterated for the last ten years.  It supports simplex and half-duplex asynchronous serial communication for up to 254 devices over a single wire. The maximum length of the bus can reach between 800 and 2000 meters depending on the mode used. It is a valid alternative to 1-Wire because of its flexibility and reliability. Fault tolerance schemes can be easily implemented because communication pins can be configured at runtime.  

It is so reliable you can try (at your own risk) touching the pin of an arduino with one hand, and the pin of another arduino with the other hand. Incredibly, the protocol works even through the human body, here the most basic example: https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/BlinkTest

As I told you before, it can substitute RS485 if the setup is not exposed to huge interference (industrial stuff), in the home automation environment, near AC wiring or microwave ovens it operates flawlessly. Many users have based their own home automation setup on this.

HEvNeck, 2020-06-17:
Hi gioblu,

Thanks for your fast reply again.

May i conclude from this, that you suggest to leave out the RS485 transceivers completely? And use BitBang instead?
In that case i understand that i did not see the relationship between the two strategies!

But i think that i'll go for the 485 solution in the first place, because all (except one) of the slave units are already in use at the moment, and it would require some pretty major modifications on their PCB's.

but still.... BitBang sounds very tempting, especially when i run into a lot of RS485 problems again (as once before...) We will see how matters turn...

B.t.w When you are not powering your Arduino with a Transformerless Power Supply Design, the risc of touching an Arduino would be less than Nihil! ;-)

Cheers, Hans

gioblu, 2020-06-17:
@HEvNeck yes, I was suggesting to run SoftwareBitBang instead, that does not require any additional hardware to operate. I think the most important difference between PJDL and RS485 is that multi-master mode is much more efficient if you use PJDL. So if you often need to arbitrarily send data from the slaves, PJDL will likely be more stable, have a better performance and less collisions.

HEvNeck, 2020-06-17:
I was trying to avoid the collision-problem by only letting the master take the initiative, by requesting the slaves on a polling base...
But even that is perhaps more difficult as one would think. The main problem lies in the slaves having to buffer the messages to send on request. Memory stays scarce on Arduino.. (I even thought about SPI-memory expansion for that... but that also, would require some pretty heavy rework on the PCB's)

gioblu, 2020-06-17:
@HEvNeck if you use the request-response scheme, and so operate in master-slave only, it should be stable and reliable using RS485. I suggest you to create a test setup and make a performance comparison. Many users after dropping RS485 and using PJDL are really happy and would never go back.

HEvNeck, 2020-06-18:
Hi gioblu,

Thanks for your advice. 
I was planning to re-write some of my existing code to let it make use of Pjon. 
But i think i'm going to run in to much trouble mixing this (to me...) new technology and the pretty complex code already written (more than 600 lines for the simplest one)

So I'll create a test-setup first, as you suggest, and play with the options i have.

Thanks for now! You may close this issue if you want, as i've not very much time to work on it; Just a few spread hours in the week,. So the progress will be pretty slow.

With kind regards, Hans

gioblu, 2020-06-18:
Ciao @HEvNeck sure, if you need any help or want to have a chat you can find me in our gitter chat room here: https://gitter.im/gioblu/PJON

If you run into any issue please report it :)

Thank you again for your support.

closed 2020-06-18 by gioblu