gioblu/PJON #356: I can't find Master Slave Auto Addressing Example and more...

Kaitoukid93, 2020-06-23:
I'm Using STM32f103 (blue pill) as the Master device to transmit data package to a network of
arduino nanos Using hardware serial (connect all Rx to the same blue pill's TX)
I used the "through serial" example, STM32F103 with transmitter sketch, it send 'B' to each device with 2 different ID as the sketch I show below
but the LED in pin 13 is not flash as I think it should (1 sec interval and 2 seconds interval), they're randomly flashed, I check the serial monitor on the transmitter device and letter B also randomly appear and at the really fast rate, not 1000000 and 2000000 microseconds
I also check and set the arduino nano as the master and same thing happen, the interval is not correct 
I also need the auto addressing example that I cant find out 
thanks for your help, great library,

`#include <PJON.h>

PJON<ThroughSerial> bus(45);
/* Use PJON<ThroughSerialAsync> bus(45);
   instead if non-blocking asynchronous operation is required */

void setup() {
  Serial.begin(115200);
  bus.strategy.set_serial(&Serial); // Pass the Serial object you want to use for communication
  bus.begin();
  bus.send_repeatedly(44, "B", 3, 1000000); // Send B to device 44 every second
  bus.send_repeatedly(46, "B", 3, 2000000); // Send B to device 44 every second
};

void loop() {
  bus.update();
};`

gioblu, 2020-06-23:
Ciao @Kaitoukid93 I have never tried a bus in that configuration. Does it work as expected when having a point-to-point connection between just 2 devices? What you can do is to try disabling the acknowledge and see what happens.

The auto or dynamic addressing has been moved into the [OSPREY](https://github.com/gioblu/OSPREY) repository although is still in unstable state.

Kaitoukid93, 2020-06-23:
Thanks @gioblu , I tried 1, 2 ,3 devices and they act the same, the 'B' is randomly sending, not following the interval that I set
this is what shown on Serial Monitor. As I said, the data not constantly show up, they "choke" at some point and then going crazy fast, but they're a lot faster than 1 second interval. If it's 1 second then slowdown, I will think of interference or blocking code but this case, I don't think show
https://www.youtube.com/watch?v=ptnjuZbzYyY&feature=youtu.be

I want to tell you about my project because I'm new to this library and I dont know if it's possible.
So the document about the auto addressing might cover all the function that I need .This is what I have drawn before knowing about your library:
Phase 1:
-At first, only Master device (here is the STM32f103) has the ID, then when new device(s) is connected (no matter how many device connect at the same time or one by one (of course number is smaller than 254)).
- The master will ask the slave(s) to return their Long ID (rID) and short ID(ID that smaller than 254), this ID is unique because I think that I will pre-program it since I dont need to make more than 4 billions devices.
-If the ID is 0(0 is the default(not assigned  slave ID)) then the master will auto generate the ID for that device acording to it's rID
-after this Phase, all the slaves will have their own ID and rID, every power cycle, this phase will repeat.

Phase 2: Since I got all the ID(that use for transmit data), now I send the packets with different payload to control all the slaves independently.

What I'm confusing now :
Will the Pjon through serial works fine with half duplex  because I only use 1 wire to transmit and receive (of course with some hardware tweaks)
If the serial does not meat the requirements so can I use SBB at higher speed to archive  115200 baud rate
There are more question to come but this is what confuse me the most.
Sorry I'm noob

Kaitoukid93, 2020-06-23:
> Ciao @Kaitoukid93 I have never tried a bus in that configuration. Does it work as expected when having a point-to-point connection between just 2 devices? What you can do is to try disabling the [...]

After set ACK to false, the time interval was perfectly correct, the garbage character still there but they didn't affact the time interval, I think it's just the bus ID byte or something

gioblu, 2020-06-23:
Ciao @Kaitoukid93 the only single wire communication supported for now is with the [SoftwareBitBang](https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang) strategy.

What you see iin the serial monitor is not garbage but additional meta-data transmitted along with the payload by the PJON library. I am not sure why the acknowledgement does not work for you. What could be required is to set a higher `TS_RESPONSE_TIME_OUT` if you can give it a try defining it before the library inclusion with a value of 50000 and see what happens.

closed 2020-10-05 by gioblu