gioblu/PJON #359: SWBB dynamic handshaking

theo1cardoso, 2020-07-20:
Hello @gioblu,
I and two other colleagues are working on a project of MIDI (or equivalent) modular controller based on Arduino Uno boards.

Peripheral Arduino boards (slaves) have to send their sensors values (buttons, potentiometers, faders) to a central Arduino board connected to the computer (master), which finally transmits the values trough it serial port.
In order to reduce bandwidth use, we choose to only transmit when values change.
But we also want to allow hotplugging of peripheral boards and so we would like central board (master) to know when a peripheral board is connected or not even if it does not transmit any values.

In order to do this, we tried (see the sketchs belows)
- peripheral send a packet every time interval called "paquetCoucou" (yes we are french haha !)
- central board checks if it received the packet or not and report it in the serial port
Here is only the sketch of one of the peripherals but they all work the same.

We are observing average results with 3 peripheral and one central board. 
Sometimes the central board declares a peripheral as gone even though it is not.

First we would like to know if it sounds like a good strategy according to you ?
Second, we use the bus.send_packet function, so we thought that maybe it was because of a random collision of the different peripheral packets "paquetCoucou". So we tried to use result = bus.send_packet to send the packet again if result is not a PJON_ACK. But it doesn't seem to improve the results.
Are we missing something ?

Thank you in advance for your help.
See you around,

PERIPHERAL BOARD CODE

`/*
   Works with central
*/

#define N 8 //amount of faders

#define pinPJON 12
#define PJON_INCLUDE_SWBB
#define SWBB_MODE 4
#include <PJON.h>

const int id = 1; // Slave Id
int idM = 127; // Master Id
const char type = 'D'; // Slave Type

unsigned long previousMillis = 0;
unsigned long interval = 100L;
//----------

PJON<SoftwareBitBang> bus(id);

//Multiplexer Address Pin
const byte PIN_ADRR_A = 2 ;
const byte PIN_ADRR_B = 3 ;
const byte PIN_ADRR_C = 4 ;

//Multiplexer Signal Pin
const byte PIN_SIG = A0;

int i;
uint8_t val[N], oldval[N];
uint8_t paquetCoucou[3] = {1, type, id};

//===========================================

int readAnalogMux(byte channel)//Multiplexer Reading Function CD4051
{
  digitalWrite(PIN_ADRR_A, bitRead(channel, 0));
  digitalWrite(PIN_ADRR_B, bitRead(channel, 1));
  digitalWrite(PIN_ADRR_C, bitRead(channel, 2));
  return analogRead(PIN_SIG);
}

//========================================

void setup()
{
  //CD4051
  pinMode(PIN_ADRR_A, OUTPUT);
  pinMode(PIN_ADRR_B, OUTPUT);
  pinMode(PIN_ADRR_C, OUTPUT);
  digitalWrite(PIN_ADRR_A, LOW);
  digitalWrite(PIN_ADRR_B, LOW);
  digitalWrite(PIN_ADRR_C, LOW);
  pinMode(PIN_SIG, INPUT);

  //PJON
  bus.strategy.set_pin(pinPJON);
  bus.begin();
  bus.set_communication_mode(PJON_HALF_DUPLEX);
  
  //Declaring the slave to the Master
  bus.send_packet(idM, paquetCoucou, 3);

  Serial.begin(115200);
}

//=================================

void loop(){
   
    for (i=0;i<N;i++) {
    oldval[i] = val[i];  
    val[i] = readAnalogMux(i)>>3;
    if (oldval[i]!=val[i]) {
      uint8_t packetA[3] = {(uint8_t)(type),
                            (uint8_t)(i),
                            (uint8_t)(val[i])
                           };
      previousMillis = millis();
      if(!bus.update()) // If all packets are delivered, send another                   
      bus.send_packet(idM, packetA, 3);
     }
  }

  if(millis()-previousMillis >= interval){
      previousMillis = millis(); 
      if(!bus.update()) // If all packets are delivered, send another
      bus.send_packet(idM, paquetCoucou, 3);
      uint16_t result = bus.send_packet(idM, paquetCoucou, 3);
      if (result != PJON_ACK){
         bus.send_packet(idM, paquetCoucou, 3);
      }
     }
  
  bus.receive(1000);
  bus.update();
  //delay(10);
}`

CENTRAL BOARD CODE

`/*
 * Work with sketch  peripheral
 */ 

 
#include <Arduino.h>

#define PINtransmitPJON 12
#define PJON_INCLUDE_SWBB
#define SWBB_MODE 4
#include <PJON.h>

#define SERIAL_RATE 115200

#define Nperiph 4 //amount of modules that can be connected

unsigned long interval = 2000L; //timeout beyond which we can consider the slave to be deconnected

int val, oldval;

const int idM = 127; // master id
int id; // slave id
char pac1; // First element of received packet PJON
int pac2 = 0; // Second element of received packet PJON
uint8_t pac3 = 0; // Third element of received packet PJON

unsigned long previousMillis[Nperiph];
uint8_t moduleconnect_type[Nperiph];
int moduleconnect_id[Nperiph];
bool con[Nperiph];

int Nperiph_reel = 0; //amount of realtime connected slave

char type; // Type of slave
int midiNote = 0; // MIDI note to transmit
int numPar = 0; // parameter index to transmit
int valPar = 1; // parameter value to transmit

PJON<SoftwareBitBang> bus(idM);

//===================================

void envoie_parametre_serie(int identite, int num_parametre, int val_parametre){
  Serial.print(2);
  Serial.print(" "); //=32
  Serial.print(identite);
  Serial.print(" "); //=32
  Serial.print(num_parametre); 
  Serial.print(" "); //=32
  Serial.println(val_parametre);
}

//-------------------------
void fonction_receptionPJON(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  
  if(payload[0] != 'M') {
    id = (packet_info.sender_id) ;
    pac1 = ((uint8_t)(payload[0])) ;
    pac2 = ((uint8_t)(payload[1])) ;
    pac3 = ((uint8_t)(payload[2])) ; }
   

  if (pac1 ==1){ //Packet test for handshake
    for (int i=0;i<Nperiph;i++){
      if (pac3==i){ //test to evaluate which slave sends "Connected Packets"
        moduleconnect_type[i] = pac2;
        moduleconnect_id[i] = pac3;
        previousMillis[i] = millis();
        bool oldcon = con[i];
        con[i]=1;
        if(oldcon != con[i]){
          Serial.print(1);
          Serial.print(" ");
          Serial.print(moduleconnect_id[i]); 
          Serial.print(" "); //=32
          Serial.println(moduleconnect_type[i]);
        }
      }
    }
  }
  else{
    
    switch (pac1){  //Evaluate the type of the slave
      case 'A' : midiNote = pac2 ; break;
      case 'B' : 
                type = pac1;
                numPar = pac2 ;
                valPar = pac3;
                envoie_parametre_serie(id, numPar, valPar);
                previousMillis[id] = millis();   
               break;
      case 'C' : 
                type = pac1;
                numPar = pac2 ;
                valPar = pac3;
                envoie_parametre_serie(id, numPar, valPar);
                previousMillis[id] = millis();   
                break;
      case 'D' : 
                type = pac1;
                numPar =  pac2;
                valPar = pac3;
                envoie_parametre_serie(id, numPar, valPar);
                previousMillis[id] = millis();    
                break;
    }
  }  
}

//===================================================================

void setup(void) {
  Serial.begin(SERIAL_RATE);

  for(int i=0;i<Nperiph;i++){con[i]=0;}
  
  bus.strategy.set_pin(PINtransmitPJON);
  bus.begin();
  bus.set_receiver(fonction_receptionPJON);
  bus.set_communication_mode(PJON_HALF_DUPLEX);
}

//==================================================================
void loop(void) {

 if (Serial.available()>0){
  if (Serial.read() == 60){
    for(int i=0;i<Nperiph;i++){con[i]=0;}
  }
 }
 
  for (int i=0;i<Nperiph;i++){
    if (millis()- previousMillis[i] >= interval){  // Deconnexion test to evaluate wether the PJON Slave is switched off
      previousMillis[i] = millis(); //resets the counter
      if(moduleconnect_id[i]!=0){
        bool oldcon = con[i];
        con[i]=0;
        if(oldcon != con[i]){
          Serial.print(0);
          Serial.print(" "); //=32
          Serial.print(moduleconnect_id[i]); 
          Serial.print(" "); //=32
          Serial.println(moduleconnect_type[i]);
        }
      }
    }
   }
  
  bus.receive(100000);
}`

gioblu, 2020-07-20:
Ciao @theo1cardoso welcome :)
Cool project!

Please fix a little the code indentation, and make ALL the Serial prints optional. When you transmit serial and SoftwareBitBang data in parallel (at the same time) SoftwareBitBang's performance drops. I think performance will be nominal if you remove the serial transmission. If you effectively need to send both serial and SoftwareBitBang data, after caling `Serial.print` call `Serial.flush` which should block until serial data is transmitted. I see you are using the `send_packet` function, consider that has no guarantee of success, use `send_packet_blocking` instead, it will retry for up to 3.5 seconds. If you use `send_packet` or `send_packet_blocking` you don't need `update`, that must be called only if you use the `send` or `send_repeatedly` functions (which are non blocking and make use of the packet buffer).

theo1cardoso, 2020-07-21:
Ciao @gioblu,
Thank you for your quick response !

We are going to clean the code and try these different things today.
But we would like to know more about Serial and SoftwareBitBang concurrency. 
Given that we need central board to communicate with computer, it is required that we use a protocol available on this board. We are using Arduino Uno so serial is, if I'm not mistaking, the only protocol if not using an extra shield.
So is there a wire PJON strategy (half-duplex mode) that allows to use Serial duplex transmission in parallel smoothly ?

Thank you in advance,

gioblu, 2020-07-21:
Ciao @theo1cardoso for parallel or concurrent PJDL and Serial communication (that is a problem) I mean:
```cpp
Serial.print("CIAO");
bus.send_packet(44, "CIAO", 4);
```
In this particular case, Serial data is transmitted while PJON is transmitting data. In the example below instead serial data is transmitted, and when that is finished PJON data is transmitted.

```cpp
Serial.print("CIAO");
Serial.flush(); // Block until Serial data is transmitted
bus.send_packet(44, "CIAO", 4);
```

This avoids performance drops.

gioblu, 2020-07-21:
@theo1cardoso forgot to say, try to see if you can get away with master-slave, that approach would be much easier to implement than multi-master specially if you are making the first steps with the library. With this I mean, let the master decide the pace at which data is gathered through a request (from master) and a consequent response (from the slave) and avoid, if you can, slaves to arbitrarily transmit data without being asked. If you need to route data from the master to the computer over serial do that after you gathered all the data you need from the slaves, wait the serial transmission is completed and loop back.

theo1cardoso, 2020-07-22:
Bonjour @gioblu,
Thank you for your responses.

We make a little break and will work on the project again on Friday.
We will try to implement a master-slave approach.
However given that we might have a certain amount of peripherals, we fear that if slave wait for a request from master before sending data, sensor values received by master won't be fluent.
We will in the same time try to separate PJON and Serial transmission as you advised.

Keep you in touch,

closed 2021-03-25 by gioblu