gioblu/PJON #362: feature-request: It is possible to create a PJON_Receiver to listen on a specific port?

Cimex97, 2020-08-25:
Hey,
I want to implement two applications that use the same PJONBus or PJONRouter.
It is possible to set for a port a PJON_Receiver function?
For example: 

```
bus.set_receiver(pjon_receiver_cb, port);
```
I know that I can use this function:
```
bus.include(true, port);
```
But this works only for one port and I won't get other messages.

Some background information:
I have a system with a couple of devices using PJON. This hardware runs with the [zephyr-rtos](https://www.zephyrproject.org/). On the systems runs also the [MCUBoot Bootloader](https://mcuboot.com/) that I can upgrade the firmware of the devices remotely. This is done by the SMP(Simple Management Protocol) that comes with the bootloader.

My Idea is to add a port for the SMP and "route" the packets to the zephyr backend, otherwise the application get the packets.

Something like that:
[image]

Is this an use case for the PJON protocol?

gioblu, 2020-08-25:
Ciao @Cimex97 yes that looks a potential use case for the PJON protocol. Look at the implementation of the setter:
https://github.com/gioblu/PJON/blob/12.1/src/PJON.h#L815

It is made to let you call `include_port(true)` this sets the port as `PJON_BROADCAST` then:
https://github.com/gioblu/PJON/blob/12.1/src/PJON.h#L391
because of that, received packets are never dropped whatever is the value of `port`.

if you set `include_port(true)` you can then add a couple of conditionals in the receiver function where to choose what to do depending on the port. 
```cpp
void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  Serial.print("Packet received, ");
  if(packet_info.header & PJON_PORT_BIT) {
    if(packet_info.port == 8001) Serial.print("SMP");
    if(packet_info.port == 9001) Serial.print("APP");
  }
  Serial.println();
}
```
The only difference is that you need to use one of the send functions specifying which port to use for each packet transmission, for example (from [docs](https://github.com/gioblu/PJON/blob/12.1/documentation/data-transmission.md)):
```cpp
// All optional parameters available
bus.send_packet(
  10,         // Device id (uint8_t)
  "Hello",    // Payload   (const void *)
  5,          // Length    (uint16_t)
  bus.config, // Header    (uint8_t)  - Use default config
  0,          // Packet id (uint16_t) - Don't include packet id
  8002        // Port      (uint16_t)
);
```

Let me know if help is required :) it would be cool to see the thing working!

gioblu, 2020-08-25:
Ciao @Cimex97 I will document this feature better in v13.0 documentation, thank you for exposing the lack of clarity about this in the documentation.

Cimex97, 2020-08-26:
Thank You for the answer. I will try it out and let you know if it works.

Cimex97, 2020-09-11:
Hey, 
I implement a simple PJONSession class for me to handle this. I can now upgrade my devices, that run with Zephyr-OS, over PJON.
I will create a PR soon for the Zephyr interface.

closed 2020-09-11 by Cimex97