gioblu/PJON #369: ESPNOW and MAC Address Sending

porkyneal, 2020-10-18:
Many thanks for your great library, got up and running very quickly with two ESP32s.
My problem now comes when I am trying to talk between 3 or more ESP32s

I am using #define PJON_INCLUDE_MAC

My aim is to only use Mac Addresses to communicate with specific devices, rather than the PJON ID, is this possible?

```
  PJON_Packet_Info info;
  info.header = bus.config | PJON_MAC_BIT;
  memcpy(info.rx.mac, mac, 6);
  if(bus.send(info, buffer, _size) == PJON_FAIL){
    DEBUG_APP("PJON failed to send...\n");
  }; 
```

I seem to have a problem where when sending to a device that was the second device to have registered, the message gets sent to the first device to be registered., But fails because the info.rx.mac does not match that device. My short debugging time looks to point at the fact all the devices have an ID of 255.

``` 
        int16_t pos = find_remote_node(id);
        if(pos != -1)
            en.send_frame(data, length, _remote_mac[pos]);
``` 

I think find_remote_node always returns position 0, which then uses the MAC address that registred first.

Any pointers would be greatly appreciated.

porkyneal, 2020-10-18:
For now, I have used bus.set_id(random(1, 254)); to get round this issue. 
But I am hoping to use all 10 device, and cannot risk a clash. 
Addtionally, hardcoding of IDs is not a route I would like to take.

gioblu, 2020-10-18:
Ciao @porkyneal yes, you are right, thank you for raising this. The ESPNOW strategy and in particular `find_remote_node` are not designed to handle mac addressess. It looks `find_remote_node` should receive a second `mac` parameter. In `send_frame` the strategy needs to parse the packet to get the mac address if present within the payload and pass it to `find_remote_node`, the function should use a new space in the buffer if the id is the same but the mac addresses differ.

gioblu, 2020-10-18:
@porkyneal we have recently added MAC support within the PJON format this is why `ESPNOW` is in this state.
To be able to communicate to all devices with acknowledgement support my suggestion is to use `PJON_NOT_ASSIGNED` device id for all nodes, after `find_remote_node` and `send_frame` are patched it should work fine. Let me know if you want to attempt to fix the source or you want me to do it.

gioblu, 2020-10-19:
Ciao @porkyneal I have merged your pull, made few minor changes and added you in the list of contributors.
Thank you again for your support.

porkyneal, 2020-10-19:
No worries, appreciate the quick responses!

closed 2020-10-19 by porkyneal