gioblu/PJON #374: Raspberry Pi as master with RS485

akadlec, 2020-12-14:
Hi to all,
i have a small network composed of arduino as a slaves and one esp8266 as a master. Using TS over RS485. This setup is working great. But right now I would like to replace esp with rpi with rs485 header. My idea is to use rpi to send and receive messages and proxy them to some message queue to be handled by my PHP app.

Is it good or bad idea? Did someone similar setup?

akadlec, 2020-12-15:
Or is better idea to have anoter arduino as a proxy, where will be "master" sketch for the pjon network and rpi connected via serial link to aurduino?

gioblu, 2020-12-15:
Ciao @akadlec ThroughSerial over RS485 works fine on RPI as well, it should work fine with the setup you describe!

akadlec, 2020-12-15:
Hi @gioblu thanks for your reply. I will try to start with an example code from this repo. And what do you think about the rs485 and TS? I saw in other issues your comments where you recommend other strategy.

gioblu, 2020-12-15:
I suggest you to try SoftwareBitBang, with a single wire you can connect up to 255 devices and it supports multi-master. 
The only con is that you need a TS-SWBB switch between nodes and the RPI (SWBB does not run on any rtos because of timing issues).

closed 2025-06-01 by gioblu