gioblu/PJON #375: Node-Red support

PizzaProgram, 2020-12-15:
I've accidentally found this project and I'm amazed by it at first view.
Spent 600 working-hours during last 1 year to search for a : 
- stable
- auto-reconnecting
- hot swap capable
- addressable / "named" boards
- minimal wiring but NO wireless!
- with long distance (5-50m) capability
standard, that could help me connect some Tiny85 and Arduino Nano boards to my Raspberry Pi to control my home.

I've also learned after 8 month, that 75% of the people who wants to do the same turn to [NodeRed](https://nodered.org/) .
But currently there is NO STABLE SOLUTION to connect all kinds of these things!
- Serial connection examples have no parity/CRC + overflow + reconnect control
- Firmata can not easily [identify boards by name/serialnumber](https://github.com/firmata/arduino/issues/471)
- Couldn't decipher how [Ardulink](http://www.ardulink.org/) would do that job
- [Serial2MQTT](https://vortex314.github.io/serial2mqtt.html) is a one-man-project not willing to standardize 

# Would it be possible to use PJON from Node-Red?

gioblu, 2020-12-15:
Ciao @PizzaProgram for sure SoftwareBitBang strategy fits in your requirements, some users did use PJON along with Node-Red, although we have not prepared a ready made example.

Consider that PJON runs also on a real operative system, so you could:
- have an instance of PJON over SoftwareBitBang running in each attiny/arduino
- have an instance of PJON over SoftwareBitBang and one PJON over ThroughSerial in a device that is connected to an rtos via usb (yet another arduino acting as a SoftwareBitBang<====>ThroughSerial switch)
- have an instance of PJON over ThroughSerial in the rtos that communicates to node red using its API

PizzaProgram, 2020-12-16:
( _Thanks for the quick answer! I was investigating everything during the last 14 hours before answering._ )

- It's not clear yet, which one would be the best solution. With ThroughSerial I'm back at the beginning. 
  (_What happens if the COM port is taken / can not reconnect / changes from "..USB0" to "USB1"_ ... )
  
- I've thought MQTT would be best solution, but I don't see how, if connected with USB to the RPi.
  ( _Normally in my thought there would be a gateway program that converts PJON protocol to MQTT messages running on RPi?_ )
   
- What I still not find is the "server / service" program that can be installed to the RPi side, as master to handle the incoming PJON slaves.

- Also it's not clear looking at the example codes, why is  `Serial.begin(9600);` used? Is that the max speed?

- Is it possible to give NAMES to the firmwares? Or Device-ID (+ MAC) only ?

- `SWBB_MODE = 4` is the max speed? ( _If using 5m/1.8m good quality shielded USB cable._ ) 3.34kB/s seems to be too few.
  To quickly transfer multiple analogue values measured from multiple boards while still need to responding to IO ports immediately, how many bits gets totally transferred?
  8 analogue (x16 bit) + 10 IO states + CRC8 + overhead  = 150  x every 0.1 sec = 1500baud / board 
  That's **max 18 Arduino NANO / bus** if used all ports. Right?
  

Thank you!

# Why so many questions?
I'm sharing my thought to show how a "newcomer" sees and searches things first visiting your project.
It may help to build a FAQ page from my questions later.

_PS: Of course I'll donate if it works. (I like if people pay for me for my help > so it's obvious I usually do the same) ._

PizzaProgram, 2020-12-16:
I've also found this:

https://github.com/Halytskyi/PJON-gRPC/tree/master/examples/devices/two_routers
Which seems to be a bit overkill to waste +2 boards just as gateways, but promising.

And there is a gRPC node for Node-RED:
https://flows.nodered.org/node/node-red-contrib-grpc

gioblu, 2020-12-16:
Ciao @PizzaProgram 

- You can run 2 instances of PJON in parallel, for example EthernetTCP and SoftwareBitBang, or MQTTTranslate and SoftwareBitBang
- You could use one of the MCUs supported by SoftwareBitBang which can also run the MQTTTranslate strategy (for example the ESP8266) you then just need to implement a switch
- What you would need the "server" if you use MQTTTranslate https://github.com/gioblu/PJON/blob/master/examples/WINX86/Local/MQTTTranslate/PingPong/Receiver/Receiver.cpp really looks like what you program in the "client" as you can see the library can be compiled on x86 as well as on all other targets, so the use of the library is pretty much the same everywhere
- 9600Bd is the default speed used by the Arduino IDE and its examples I chose to conform to that and let users then set a faster baudrate if required, I have tested it at 1MBd Arduino-RPI and it worked not sure at higher speeds
- `SWBB_MODE = 4` is the fastest mode of [SoftwareBitBang](https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang), which is an implementation of [PJDL](https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/specification/PJDL-specification-v5.0.md) or a single wire communication protocol I have designed in the last decade with the help of few friends, unlike serial it can cross 2km of twisted pair, and up to 255 devices can share the same pin for communication along with a ground connection, it is slower than serial because it is optimized for a different use case, depending on the MCU you use you could achieve higher speeds at the cost of a lower maximum range.

gioblu, 2020-12-16:
@PizzaProgram see also https://www.youtube.com/watch?v=GWlhKD5lz5w

I agree with you that https://github.com/Halytskyi/PJON-gRPC may be a quick solution to link node red with PJON.

Thank you for your support, I will for sure make use of your valuable feedback :)