gioblu/PJON #377: Clarification on what is the "padded" and the "jittering" aspect of PJON. Also maybe differentiation between UART and PJON data link layer.

mofosyne, 2020-12-27:
https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/specification/obsolete/padded-jittering-protocol-specification-v0.1.md

Was doing a quick read of the above and while I think I get the general idea of the spec. I'm not too sure I understand all the design decision behind this spec. Is there a list or page explaining the reason it's designed that way?

Also it would be nice if there is an explanation as to what PJON data link layer is standing for. As in what is the "padded" and the "jittering" aspect of this standard. As well as a reason why to use this byte framing rather than the very common UART style of framing.

While it won't be directly tied to implementation, it would at least help explain to devs why its going done that way. E.g. The padded bit is to allow for variable clock rate between bytes and device clocks.

gioblu, 2020-12-27:
Ciao @mofosyne the spec you link is the first iteration dated 2010, the updated spec is found here:

PJON (Padded Jittering Operative Network)
https://github.com/gioblu/PJON/blob/master/specification/PJON-protocol-specification-v4.0.md

PJDL (Padded Jittering Data Link)
https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/specification/PJDL-specification-v5.0.md

You see from the PJDL spec we have a synchronization pad used to re-sync every 8 bits. In between a frame transmission and a response the transmitter makes the bus "jitter" to avoid other devices to start transmitting and break the exchange.

gioblu, 2020-12-27:
@mofosyne about the differences between PJDL and UART:
- PJDL can reach 2km of range using a twisted pair
- PJDL supports up to 255 devices connected to the same data and ground wires
- PJDL transmits also the clock signal along with data
- PJDL supports the acknowledgement 
- PJDL is more interference-proof 

Upon PJDL generally is used the PJON network protocol that is more comparable to IP

closed 2021-01-02 by gioblu