gioblu/PJON #378: reduce memory footprint of PJON

level20peon, 2020-12-29:
I am trying to use PJON in order to establish communication between a Raspberry Pi and a couple of ATTiny85s. In order to do so, I am using an Arduino as router, which communicates with the Pi via Serial Link and with the Tiny via BitBang.

Simple examples like sending a message from the Pi to the Tiny and the Tiny replying are working. However, even if I only create a simple "Blink LED on any message that is received" sketch for the Tiny and only including PJONSoftwareBitBang.h without including any other library, the RAM usage on the Tiny is at over 80%. When I throw in a few algorithms the RAM usage easily exceeds 90%. This makes the Tiny very unstable (crashing, most of the time without recovering unless I pull the plug).

So, is there a way to shed off some of those 80%? I only need to receive and send very small text strings, 10 characters would suffice. Can I throw out some optional functions? Can I maybe use smaller variable sizes? I already tried putting the consts into PROGMEM and using compiler constants rather than constant variables but that didn't change the RAM usage on the Tiny at all.

Any pointers?

gioblu, 2020-12-29:
ciao @level20peon, welcome, here some ways to reduce the footprint:
- Use only `send_packet` function, don't use `send`, `send_repeatedly` or `send_packet_blocking`
- Reduce the length of the packets defining `PJON_PACKET_MAX_LENGTH` before the inclusion
- Reduce the length of the packet's buffer defining `PJON_MAX_PACKETS` with a value of `0` before the inclusion
- Do not call the `update` function

Doing so you should bring the ram usage down to 100 bytes.

level20peon, 2020-12-30:
Wow, thank you @gioblu!

Actually I noticed the following:

Changing `PJON_PACKET_MAX_LENGTH` to `10` and `PJON_MAX_PACKETS` to `0`  without changing anything else in my code (so I still had `send` and `update` in there) brought RAM usage down to 88 bytes.
**But** using `send_packet` instead of `send` actually increased RAM usage to 100 bytes.
Also, using `update` (vs. not using it) didn't change RAM usage at all. I am only using it in the loop, so maybe that's why it doesn't make a difference for me?

gioblu, 2020-12-30:
ciao @level20peon no problem, it's a pleasure to help 👍 
It uses less ram when using `send` because there is no buffer in memory, although it would not work.
Stick with `send_packet` if you can, if you need guaranteed delivery and retries use `send_packet_blocking` instead

gioblu, 2020-12-30:
@level20peon if you use PJON In local mode and you need to send payloads of 10 bytes, use `PJON_PACKET_MAX_LENGTH` with a length of 15 (10 for the payload and 5 for the overhead of PJON)

level20peon, 2021-01-02:
ok, I'm still experimenting a little bit with packet delivery optimization, but the goal of actually reducing the memory footprint is solved. So I will close this topic and open other issues as they arise, because I don't want to mix topics in a single thread.

Thank you again 👍

closed 2021-01-02 by level20peon