gioblu/PJON #379: packet_info.tx.id is always empty in PJONThroughSerial

level20peon, 2021-01-02:
I am trying to print the packet_info.tx.id on my receiver but it seems to always be empty. My sender tries four different approaches, none of them lead to my receiver being able to print the sender id. What am I doing wrong?

Expected output:
> Header: (no idea what would be displayed here) Sender id: 3 Length: 6

Actual output: 
> Header:  Sender id:  Length: 6

**Sender**
`#include <PJONThroughSerial.h>

#define idTS 3

PJONThroughSerial busTS(idTS);

void setup() {
  Serial.begin(9600);
  busTS.strategy.set_serial(&Serial);
  busTS.begin();

  // send "foobar" to device id 1
  busTS.send(1, "foobar", 6, busTS.config);

  // send "foobar" to device id 1 including "busTS.config"
  busTS.send(1, "foobar", 6, busTS.config);

  // send "foobar" to device id 1 including "PJON_Packet_Info"
  PJON_Packet_Info info;
  info.rx.id = 1;
  info.tx.id = 3;
  busTS.send(info, "foobar", 6);

  // send "foobar" to device id 1 including "busTS.config" and "PJON_Packet_Info"
  PJON_Packet_Info info;
  info.rx.id = 1;
  info.tx.id = 3;
  busTS.send(info, "foobar", 6, busTS.config);
};

void loop() {
  busTS.update();
  busTS.receive();
};
`

Receiver
`#define idBB 1
#define TS_TIME_IN 0	// Minum duration of channel free for use before transmission - set to 0 because of Master-Slave mode

#include <iostream>
#include <wiringPi.h>
#include <wiringSerial.h>

#ifndef RPI
	#define RPI true
#endif

#include <PJONThroughSerial.h>

PJONThroughSerial bus(idBB);

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  std::cout << "Header: ";
  std::cout << packet_info.header;
  // If sender device id is included
  if(packet_info.header & PJON_TX_INFO_BIT) {
    std::cout << " Sender id: ";
    std::cout << packet_info.tx.id;
  }

};

int main (int argc, char* argv[]) {
	uint32_t baud_rate = 9600;
	printf("PJON ThroughSerial BlinkWithResponse Transmitter test \n");
	printf("Opening serial port /dev/ttyAMA0 @ %d \n", baud_rate);
	int s = serialOpen("/dev/ttyAMA0", baud_rate);
	if(int(s) < 0) printf("ERROR - Serial open fail");
	if(wiringPiSetup() == -1) printf("ERROR - WiringPi setup fail");
	printf("Strategy Serial link configuration \n");
	bus.strategy.set_serial(s);
	bus.strategy.set_baud_rate(baud_rate);
	printf("Bus initialization \n");
	bus.set_acknowledge(false);
	bus.set_receiver(receiver_function);
	bus.begin();

	while(true) {
		bus.update();
		bus.receive();
	}

	return 0;
};
`

fredilarsen, 2021-01-02:
Is this the exact code? I see no printout of "Length" anywhere.

The id is a uint8_t so it cannot be "empty". If it gets printed as empty it must the stream that handles it incorrectly -- can you try a typecast, for example `cout << (int) packet_info.tx.id;`?

level20peon, 2021-01-02:
I snipped a few lines too much when posting, so I added the "Length" above.

The typecast with `(int)` you proposed worked, thank you! I will adapt this to my current code and report back if everything I subsequently intend to do with that id works, too.

level20peon, 2021-01-03:
ok, everything works as intended, thanks for your help! 👍

closed 2021-01-03 by level20peon