gioblu/PJON #382: PJON + PJON Cython + TSA

akadlec, 2021-01-04:
Hi @gioblu
right now i'm moving my master code to python with pjon_cython package. Everything is working fine, but with a small issue. Master is resending packets :/ It looks like when acknowledge function is enabled there is some issue with timing and master is maybe to fast? Do you have any advice?

gioblu, 2021-01-10:
Ciao @akadlec sorry for the very late answer, I must have missed this.
If master re-transmits packets you may try setting `TSA_RESPONSE_TIME_OUT` to a higher value than default (45000), that should help. I suggest you to also switch to PJON v13.0 TS which is an updated version of TSA.

Let me know how it goes, and sorry again for the late response.

akadlec, 2021-01-10:
Hi @gioblu, thanks for you reply. Right now I'm not able to configure timeout :/ on rpi I'm using `PJON_cython` and this port does not support this configuration :(

akadlec, 2021-01-26:
Hi @gioblu
So I did some other tests on my dev env. Just to be clear, my master is RPi with `pjon_cython` v12 using TSA and right now one slave is Uno with v12 too.

So I had still some issues with message sync or async ack a got a lot of `PJON_Connection_Lost` states. So I tried to change baud rate from 115200 to 38400 and result is better. Sync ack is now working without any error.

The parameter `TSA_RESPONSE_TIME_OUT` is set to `100000` by default in PJON_cython and can not be changed in runtime.

And also I did some tests with v13 on both sides with same results, everything is working for lower baud rate.

gioblu, 2021-01-28:
Ciao @akadlec thank you very much for your report.
In this iteration I will work to enable quicker baud rate without errors.