gioblu/PJON #385: ARM Cortex-M3 support?

canDry, 2021-02-10:
I'm trying to get PJON working on an arduino DUE (ARM Cortex-M3).   
  
Although there's no errors in compilation the network test never connects with ans ESP32 receiver (it works successfully when the sender is an arduino uno on pin 12).  I've tried various pins (12, 25, 52) with no luck.  
  
Is there support for this board?  
  
Cheers!

gioblu, 2021-02-11:
Ciao @canDry :)
Take a look here: https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/Timing.h#L207
This is the timing configuration of ESP32, we do not have a dedicated setup for Arduino DUE, although if you already see even few successful exchanges that means a slight tweak will likely yield optimal performance.

Please see: https://github.com/gioblu/PJON/wiki/SoftwareBitBang-troubleshooting

canDry, 2021-02-11:
> Take a look here: https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/Timing.h#L207

Thanks for the quick response Gio!   
  
Was hoping perhaps someone will see this that's got the timing working (but never bothered to pull request).  😎

gioblu, 2021-02-12:
Ciao @canDry it is probably the time for me to buy an Arduino DUE, I will let you know how my tests go.

canDry, 2021-02-12:
I have a cheap DSO-138 oscilloscope....  do you think that would be sufficient for me to diagnose/tune the timing?  
  
Regardless, if you actually purchase a DUE to solve this issue and you fall into the _relatively-poor open-source-developer camp_ send me a copy of the bill and I'll reimburse you via paypal.  :)  
  
Cheers!

gioblu, 2021-02-13:
@canDry I don't think you need an oscillo to find the right timing it you already get some packets through. Try changing `SWBB_READ_DELAY` on due side and see if results are better, it may be just that setting a slightly higher or lower `SWBB_READ_DELAY`, in the worst case `SWBB_BIT_WIDTH` may require to be tuned too, and in that case an oscillo may help, probably the DSO-138 is more than enough.

gioblu, 2021-02-13:
Ah @canDry about the DUE purchase, don't be preoccupied it is some time I had in mind to test that MCU, and that is not a huge expense, just too many things on the list :)

gioblu, 2021-02-13:
@canDry I bought also the chinese ebay clone for less then 15 dollars, curious to know if the behave the same. Do you have the real or the clone?

canDry, 2021-02-13:
> @canDry real or the clone?  
  
Mine is from Sainsmart - so a clone.  
  
If I get some time this weekend, I'll try playing with the `#define SWBB_READ_DELAY` +/-  to see if I can get the DUE talking to the ESP32.  
  
Thanks again for your prompt and very helpful response (as usual!).

gioblu, 2021-02-28:
Ciao @canDry it's a pleasure. Have you found the optimal configuration? Do you need any help?

gioblu, 2021-03-11:
Ciao @canDry I finally have my Due and some time to try! I will let you know in the next days.

fletchjp, 2022-05-13:
Is there any progress on this? I have connected two DUEs using SoftwareBitBang and get no connection. I cannot see anything in the Timing.h file which relates to a DUE. I don't know enough to know what to add.