gioblu/PJON #392: Compiler error/warning:  memset clearing an object of non-trivial type 'struct PJON_Packet_Info'

rainerschoe, 2021-04-02:
Hi,
I just ran across this warning:
```
In file included from ...../PJON/src/PJON.h:62,
...../PJON/src/PJONDefines.h: In static member function 'static void PJONTools::parse_header(const uint8_t*, PJON_Packet_Info&)':
....../PJON/src/PJONDefines.h:397:33: warning: 'void* memset(void*, int, size_t)' clearing an object of non-trivial type 'struct PJON_Packet_Info'; use assignment or value-initialization instead [-Wclass-memaccess]
  397 |     memset(&info, 0, sizeof info);
      |                                 ^
...../PJON/src/PJONDefines.h:212:8: note: 'struct PJON_Packet_Info' declared here
  212 | struct PJON_Packet_Info {
      |        ^~~~~~~~~~~~~~~~
```
As I compile my projects with `-Werror` to enforce clean code-style, this causes the build to fail. Not sure if memset is used deliberately here. in modern C++ you can just default initialize member variables
- In declaration of a struct by writing `struct S { int member = 0; };`
- In constructor of a struct by writing `struct S { S() : member(0) {} int member; };`

For now I just disables `-Werror` in my project, but it would still be nice to compile without any warnings :)

gioblu, 2021-04-02:
Ciao @rainerschoe thank you very much for reporting this. I will fix it as soon as possible.

gioblu, 2021-04-02:
Ciao @rainerschoe looking at the definition of `PJON_Packet_Info` all values are already defined as in your first proposed solution, so it looks I could just remove line 297. I will run some tests.

gioblu, 2021-04-05:
Solved, thank you for your report @rainerschoe

closed 2021-04-05 by gioblu