gioblu/PJON #394: ESP32 ThroughSerial Low Packet Count

osman-aktepe, 2021-05-28:
Hi @gioblu 

As you know, I already used your project and this time I am here for ESP32.
Before starting, I really appreciate your effort.

My environment: 2 ESP32 Dev board, 2 MAX485 module.
I wanted to test RS485 with PJON to check whether it fulfills my needs or not.

Slave Code: 
`#define PJON_PACKET_MAX_LENGTH 1024

#include <PJONThroughSerial.h>
#include <HardwareSerial.h>

// PJON object
PJONThroughSerial bus(45);
HardwareSerial HSerial(2);

void setup() {
  
  HSerial.begin(115200);
  bus.strategy.set_serial(&HSerial);
  bus.set_crc_32(true);
  // Set enable pins
  
  bus.strategy.set_enable_RS485_pin(5);

  bus.begin();

  Serial.begin(115200);
};

void loop() {
  bus.receive();
};`

Master Code:

`#define PJON_PACKET_MAX_LENGTH 1024

#include <PJONThroughSerial.h>
#include <HardwareSerial.h>

float test;
int busy;
int fail;

int bus_id = 44;
PJONThroughSerial bus(44);
HardwareSerial HSerial(2);

int packet;
uint8_t content[] = "0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789"; 
int packageSize = 10;

void setup() {

  HSerial.begin(115200);
  bus.strategy.set_serial(&HSerial);
  bus.set_crc_32(true);
  // Set enable pins
  
  bus.strategy.set_enable_RS485_pin(5);
  bus.begin();
  
  Serial.begin(115200);
  Serial.println("PJON - Network analysis");
  Serial.println("Starting a 1 second communication test..");
  Serial.println();
}

void loop() {
  long wholeTime = 0;
  long time = millis();
  while(millis() - time < 1000) {

    /* Here send_packet low level function is used to
    be able to catch every single sending result. */

    PJON_Packet_Info info;
    info.rx.id = 45;
    long startMillis = millis();
    unsigned int response = bus.send_packet(info, content, packageSize);
    wholeTime += (millis() - startMillis);

    if(response == PJON_ACK)
      test++;
    if(response == PJON_BUSY)
      busy++;
    if(response == PJON_FAIL)
      fail++;
  }

  Serial.print("Packet Overhead: ");
  Serial.print(bus.packet_overhead(bus.data[1]) + 1);
  Serial.print("B - Total: ");
  Serial.print((unsigned int)((bus.packet_overhead(bus.data[1]) + 1) * test));
  Serial.println("B");
  Serial.print("Bandwidth: ");
  // length + packet overhead + PJON_ACK
  Serial.print((unsigned int)(test * (packageSize + bus.packet_overhead(bus.data[1]) + 1)));
  Serial.println("B/s");
  Serial.print("Data throughput: ");
  Serial.print((unsigned int)(test * packageSize));
  Serial.println("B/s");
  Serial.print("Packets sent: ");
  Serial.println((unsigned int)test);
  Serial.print("Fail (no acknowledge from receiver): ");
  Serial.println(fail);
  Serial.print("Busy (Channel is busy or affected by interference): ");
  Serial.println(busy);
  Serial.print("Delivery success rate: ");
  Serial.print(100 - (100 / (test / fail)));
  Serial.println(" %");
  Serial.println("---------------------");
  // Avoid Serial interference during test flushing
  Serial.flush();

  test = 0;
  busy = 0;
  fail = 0;
};`

I changed baud rate and packet size to test different scenarios but the strange thing is, for example, when I increase baud rate from 9600 to 115200 packet sent goes from 9 to 20 but after that it does not increase. So when packet size is 20 the bandwidth is too low, around 760. I updated ESP32 base and tried several configuration changes, nothing happened. So after that, I started to increase packet size.

Here are some result;
Packet size: 20

Packet Overhead: 10B - Total: 200B
Bandwidth: 600B/s
Data throughput: 400B/s
Packets sent: 20
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 15486
Delivery success rate: 100.00 %

Packet size: 160

Packet Overhead: 10B - Total: 160B
Bandwidth: 2720B/s
Data throughput: 2560B/s
Packets sent: 16
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 5236
Delivery success rate: 100.00 %

Packet size: 960

Packet Overhead: 11B - Total: 88B
Bandwidth: 7768B/s
Data throughput: 7680B/s
Packets sent: 8
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 575
Delivery success rate: 100.00 %

As you can see, the case is strange to me and I did not understand the problem. I have checked your latest version code but before start debugging, I wanted to know your opinions.

Regards

gioblu, 2021-05-28:
Ciao @osman-aktepe with a higher packet size you implement a more efficient exchange. The overhead of each packet and its time consuming tasks (medium access control, acknowledge) are distributed on a bigger payload. In your particular case, most of the time is dedicated to medium access control, see https://github.com/gioblu/PJON/blob/master/src/strategies/ThroughSerial/Timing.h#L48 (the transmitter waits for the maximum allowed time between packet transmission and acknowledgement response to avoid disrupting an ongoing exchange)

osman-aktepe, 2021-05-28:
Hi @gioblu thats right but there is a difference; https://github.com/gioblu/PJON/issues/243, in here you can check their packet count and bandwidth with its baudrate. I think that 20 packets per second is too low. I can achieve 55 packets when I use SWBB. 

I changed my baud rate to 9600 and added these

#define TS_INITIAL_DELAY 100
#define TS_TIME_IN 10000

my packet count is 20 (by the way because of TS_TIME_IN, it is not two times better but still very low according to that case) are you sure this is normal? Do you want me to test or debug anything?

gioblu, 2021-05-29:
Ciao @osman-aktepe do you have an oscilloscope?
Could you take some pics of the output?
What was the length of the packet for the last test you speak about?

For sure something unexpected may be happening, although serial at 9600 bits per sec minus 1/5 of that time dedicated to check the line (with `TS_TIME_IN` set to 10000) is 960 bytes per sec minus the acknowledgement phase. SoftwareBitBang is more efficient and its slowest mode is 15760. 

Is the acknowledge feature a requirement for your setup? 

I suggest you to give a try to ThroughSerial disabling the acknowledge, simplex transmission is quicker and simpler, and assessing the available bandwidth in this configuration may help locating the problem, specially if you don't have the chance to probe the output with an oscilloscope.

osman-aktepe, 2021-05-29:
Hi @gioblu 

I do not have any oscilloscope, sorry :(
It was 20.

I will give some output for different baud rates and with and without acknowledge. By the way I commented out TS_INITIAL_DELAY and TS_TIME_IN so it uses default values now.

Baud Rate:9600
With acknowledge:

Packet Overhead: 10B - Total: 120B
Bandwidth: 360B/s
Data throughput: 240B/s
Packets sent: 12
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 9386
Delivery success rate: 100.00 %

Without acknowledge:

Packet Overhead: 10B - Total: 300B
Bandwidth: 900B/s
Data throughput: 600B/s
Packets sent: 30
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 0
Delivery success rate: 100.00 %

Baud Rate:19200
With acknowledge:

Packet Overhead: 10B - Total: 160B
Bandwidth: 480B/s
Data throughput: 320B/s
Packets sent: 16
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 11932
Delivery success rate: 100.00 %

Without acknowledge:

Packet Overhead: 10B - Total: 560B
Bandwidth: 1680B/s
Data throughput: 1120B/s
Packets sent: 56
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 0
Delivery success rate: 100.00 %

Baud Rate:115200
With acknowledge:

Packet Overhead: 10B - Total: 200B
Bandwidth: 600B/s
Data throughput: 400B/s
Packets sent: 20
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 15457
Delivery success rate: 100.00 %

Without acknowledge:

Packet Overhead: 10B - Total: 2000B
Bandwidth: 6000B/s
Data throughput: 4000B/s
Packets sent: 200
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 0
Delivery success rate: 100.00 %

Baud Rate:1000000
With acknowledge:

Packet Overhead: 10B - Total: 210B
Bandwidth: 630B/s
Data throughput: 420B/s
Packets sent: 21
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 16528
Delivery success rate: 100.00 %

Without acknowledge:

Packet Overhead: 10B - Total: 5000B
Bandwidth: 15000B/s
Data throughput: 10000B/s
Packets sent: 500
Fail (no acknowledge from receiver): 0
Busy (Channel is busy or affected by interference): 0
Delivery success rate: 100.00 %

It seems like acknowledge causes too much degradation. Do you need anything else?

gioblu, 2021-06-02:
Ciao @osman-aktepe sorry for the delay in the response, the acknowledge greatly reduces the available bandwidth. It operates correctly when `TS_RESPONSE_TIME_OUT` is effectively longer than the time it takes for the receiver to respond an acknowledge, if not, that is lost unheard. That is why you experience this gap in performance. What you did was the right thing to do, reduce `TS_RESPONSE_TIME_OUT` until it breaks to effectively measure what is your response time. The maximum you can do with the acknowledge at 1000000Bd is around 150/160 packets per sec, or around 5kB/s, that is not hugely bad. Leave some margin to be sure not to exceed that. As I said before this limitation is introduced by the ACK and by the latency of your hardware.

osman-aktepe, 2021-06-06:
Hi @gioblu 

Sorry for the late response, I understand what you mean but I want to  understand one thing. Basically it is timeout duration for waiting response. I do not change anything in the slave part and just change the timeout of the master. In the case that the slave send the response in a given time, normally decreasing timeout should not change the bandwidth that much, at least I would expect that. I feel like, it waits more even if it gets the response. What is the thing I am missing?

I am out of the city now but if you want me to test anything, I can test next weekend.

gioblu, 2021-06-06:
Ciao @osman-aktepe if you implement a master-slave approach over serial I don't think you need the acknowledgement. 
Master can retry if does not receive a response.

When using the acknowledgement, before transmission, master waits for a lot of time, to avoid disrupting an exchange potentially occurring. That is why you see a drop in bandwidth.

closed 2021-12-05 by gioblu