gioblu/PJON #398: What is the reason for only GPIO12 and 25 on ESP32 ?

X-Ryl669, 2021-11-19:
In the documentation, only those 2 pins are listed. I don't see anything common and specific to these pins on ESP32, so I'm wondering what is the reason PJDL would only work on such pins ?

GPIO12 is a bit painful to use since if you have it pulled high (for whatever reason) while booting, the ESP32 will likely fail to boot (it's used to change the voltage of the flash chip from 3.3V to 1.8V).
It can be pulled high if the other side of the communication link is trying to communicate while the ESP32 is rebooting. In that case, the system will never resume in a working condition, unless manually reboot **while** GPIO12 is low.

gioblu, 2021-12-05:
Ciao @X-Ryl669 I have found the compatible pins mostly by trial and error, thank you for your feedback, I didn't know about GPIO 12, that is for sure a problem, I will give a try to other pins and report to you back here.

fanczy, 2022-01-27:
Hello, any new info on this? I have an ESP32 version where pins 12 and 25 are unfortunately unavailable.

gioblu, 2022-03-19:
Thank you @X-Ryl669 for pointing this out. 
I will work to solve this ASAP.

thilojaeggi, 2023-10-20:
Any update on this?