gioblu/PJON #400: Topology - SoftwareBitBang

happen-studio, 2021-12-06:
Hi, 

I'm planning to use PJON (SoftwareBitBang) in my house, right now I'm designing some custom PCBs. 
I had a look in the docs, but I can't find any information about the wiring, what is possible what not? 

See attached a random groundplan with a couple of ESP32/custom PCB and a master (raspberrypi) to get a better picture of my question. Is this wiring possible, are there any limitations (star topology, linear topology, …) ?

The other two pictures showing a frontal perspective, with two versions of wiring. 

thanks a lot.

[image]
[image]
[image]

gioblu, 2021-12-06:
Ciao @happen-studio you can just use a single wire connecting all devices. The topology can be bus or star.
Check out the SWBB documentation here: https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang#softwarebitbang
Check out also:
https://github.com/gioblu/PJON/wiki/Protective-circuitry
https://github.com/gioblu/PJON/wiki/Mitigate-interference
https://github.com/gioblu/PJON/wiki/SoftwareBitBang-cabling-and-connectors

happen-studio, 2021-12-07:
Hi @gioblu,
thank you very much for the input, did't find the points in the wiki. ;-)
The posted groundplan is a mixture of bus and star. Is it possible to mix both topologies?

@gioblu thanks a lot for all the work with PJON.

gioblu, 2021-12-07:
Yes you can mix topologies :)
It is a pleasure :)

closed 2021-12-07 by gioblu