gioblu/PJON #403: Dead links

drtrigon, 2022-01-03:
Hi @gioblu 

Today I was browsing pjon.org and the git repo and encountered quite a number of dead links, e.g.:

- on this page https://www.pjon.org/strategies.php the links to the pages describing the stragtegies are dead
- https://github.com/gioblu/PJON/tree/master/src/strategies/OverSampling the link "PJON Oversampling packet radio" is dead

gioblu, 2022-01-15:
Ciao @drtrigon happy new year, thank you very much for your report, I will fix the links ASAP

rozrabiak, 2022-05-30:
pjon.org not working...

locriacyber, 2022-11-24:
Hello, pjon.org seem to be taken by some online gambling site

gioblu, 2022-11-24:
Ciao @locriacyber thank you for your report. I had to stop to maintain the website because I am now fully taken by work. I am obviously not affiliated with the new owners of the www.pjon.org domain. It just expired and has been bought by someone else. I will add a remark on the README just to be sure.

closed 2022-11-24 by gioblu

locriacyber, 2022-11-25:
Without offense, you don't need to maintain the website to keep the domain up.

gioblu, 2022-11-26:
Ciao @locriacyber to keep a website up means to keep up with new known vulnerabilities and standard changes that is what I meant in my answer. I will check out github pages and see if I can drop the bare HTML it in there.