gioblu/PJON #404: ESP32 to ESP32 using SoftwareBitBang and cable connection, no packets arrive

SunboX, 2022-01-05:
Hi,

I'm running a simple test, two ESP32 connected with one cable (GPIO 25 <-> GPIO 25), GND and 5V are also connected directly.

[image]

This is the code running on ESP32 "A":
```c++
#include <Arduino.h>
#include <PJONSoftwareBitBang.h>
#include <esp_wifi.h>

PJONSoftwareBitBang bus(10);

void busErrorHandler(uint8_t code, uint16_t data, void *custom_pointer)
{
    switch (code)
    {
    case PJON_CONNECTION_LOST:
        Serial.print("Connection with device ID ");
        Serial.print(bus.packets[data].content[0], DEC);
        Serial.println(" is lost.");
        break;

    case PJON_PACKETS_BUFFER_FULL:
        Serial.print("Packet buffer is full, has now a length of ");
        Serial.println(data, DEC);
        Serial.println("Possible wrong bus configuration!");
        Serial.println("higher PJON_MAX_PACKETS if necessary.");
        break;

    case PJON_CONTENT_TOO_LONG:
        Serial.print("Content is too long, length: ");
        Serial.println(data);
        break;
    }
}

void busReceiveHandler(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info)
{
    if ((char)payload[0] == 'A')
    {
        Serial.println("received A");
    }
}

void setup()
{
    Serial.begin(9600);

    btStop();
    esp_wifi_deinit();

    bus.set_error(busErrorHandler);
    bus.set_receiver(busReceiveHandler);
    bus.strategy.set_pin(25);
    bus.begin();
}

float test;
float mistakes;
int busy;
int fail;

void loop()
{
    unsigned int response = bus.receive();
    if (response == PJON_ACK)
    {
        test++;
    }
    if (response == PJON_NAK)
    {
        mistakes++;
    }
    if (response == PJON_BUSY)
    {
        busy++;
    }
    if (response == PJON_FAIL)
    {
        fail++;
    }

    Serial.print("Packet Overhead: ");
    Serial.print(bus.packet_overhead(bus.last_packet_info.header) + 1);
    Serial.print("B - Total: ");
    Serial.print((unsigned int)((bus.packet_overhead(bus.last_packet_info.header) + 1) * test));
    Serial.println("B");
    Serial.print("Bandwidth: ");
    Serial.print(test * (20 + bus.packet_overhead(bus.last_packet_info.header) + 1));
    Serial.println("B/s");
    Serial.print("Data throughput: ");
    Serial.print(test * 20);
    Serial.println("B/s");
    Serial.print("Packets sent: ");
    Serial.println(test);
    Serial.print("Mistakes (error found with CRC): ");
    Serial.println(mistakes);
    Serial.print("Fail (no data found): ");
    Serial.println(fail);
    Serial.print("Busy (Channel is busy or affected by interference): ");
    Serial.println(busy);
    Serial.print("Accuracy: ");
    Serial.print(100 - (100 / (test / mistakes)));
    Serial.println(" %");
    Serial.println(" --------------------- ");

    bus.update();

    bus.send(20, "B", 1);
    
    delay(1);
}
```

And this is the code running on ESP32 "B":
```c++
#include <Arduino.h>
#include <PJONSoftwareBitBang.h>
#include <esp_wifi.h>

PJONSoftwareBitBang bus(20);

void busErrorHandler(uint8_t code, uint16_t data, void *custom_pointer)
{
    switch (code)
    {
    case PJON_CONNECTION_LOST:
        Serial.print("Connection with device ID ");
        Serial.print(bus.packets[data].content[0], DEC);
        Serial.println(" is lost.");
        break;

    case PJON_PACKETS_BUFFER_FULL:
        Serial.print("Packet buffer is full, has now a length of ");
        Serial.println(data, DEC);
        Serial.println("Possible wrong bus configuration!");
        Serial.println("higher PJON_MAX_PACKETS if necessary.");
        break;

    case PJON_CONTENT_TOO_LONG:
        Serial.print("Content is too long, length: ");
        Serial.println(data);
        break;
    }
}

void busReceiveHandler(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info)
{
    if ((char)payload[0] == 'B')
    {
        Serial.println("received B");
    }
}

void setup()
{
    Serial.begin(9600);

    btStop();
    esp_wifi_deinit();

    bus.set_error(busErrorHandler);
    bus.set_receiver(busReceiveHandler);
    bus.strategy.set_pin(25);
    bus.begin();
}

float test;
float mistakes;
int busy;
int fail;

void loop()
{
    unsigned int response = bus.receive();
    if (response == PJON_ACK)
    {
        test++;
    }
    if (response == PJON_NAK)
    {
        mistakes++;
    }
    if (response == PJON_BUSY)
    {
        busy++;
    }
    if (response == PJON_FAIL)
    {
        fail++;
    }

    Serial.print("Packet Overhead: ");
    Serial.print(bus.packet_overhead(bus.last_packet_info.header) + 1);
    Serial.print("B - Total: ");
    Serial.print((unsigned int)((bus.packet_overhead(bus.last_packet_info.header) + 1) * test));
    Serial.println("B");
    Serial.print("Bandwidth: ");
    Serial.print(test * (20 + bus.packet_overhead(bus.last_packet_info.header) + 1));
    Serial.println("B/s");
    Serial.print("Data throughput: ");
    Serial.print(test * 20);
    Serial.println("B/s");
    Serial.print("Packets sent: ");
    Serial.println(test);
    Serial.print("Mistakes (error found with CRC): ");
    Serial.println(mistakes);
    Serial.print("Fail (no data found): ");
    Serial.println(fail);
    Serial.print("Busy (Channel is busy or affected by interference): ");
    Serial.println(busy);
    Serial.print("Accuracy: ");
    Serial.print(100 - (100 / (test / mistakes)));
    Serial.println(" %");
    Serial.println(" --------------------- ");

    bus.update();

    bus.send(10, "A", 1);
    
    delay(1);
}
```

Both are compiled and uploaded using PlatformIO with this ini file:
```yaml
[env:az-delivery-devkit-v4]
platform = espressif32
board = az-delivery-devkit-v4
framework = arduino
lib_deps = 
	gioblu/PJON@^13.0
```

The Oszilloskope is showing some communication:
[image]

But no packet is arriving at any of both ESP32. Console log:

```
 --------------------- 
Packet buffer is full, has now a length of 5
Possible wrong bus configuration!
higher PJON_MAX_PACKETS if necessary.
Packet Overhead: 7B - Total: 14B
Bandwidth: 54.00B/s
Data throughput: 40.00B/s
Packets sent: 2.00
Mistakes (error found with CRC): 0.00
Fail (no data found): 701
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %
 ---------------------
```

What's wrong? Why does no packet arrive?

SunboX, 2022-01-05:
Here are both projects to try out: https://we.tl/t-lFjg2oIFPK

SunboX, 2022-01-10:
btw., the issue number is 404 🙈

gioblu, 2022-03-19:
Ciao @SunboX I am sorry for answering you so late. I do not have the hardware to replicate your setup now, but I suspect there may be a wrong timing configuration in the esp8266. Please try to tweak SWBB_READ_DELAY and SWBB_BIT_WIDTH on the ESP8266 side (the right value may be shorter or longer, start with SWBB_READ_DELAY, if you don't see packets after tweaking it you may need to go back to its original value and try to tweak SWBB_BIT_WIDTH). Here to edit those values: https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/Timing.h#L198-L201
Try also different combinations of pins it may work better.
Let me know how it goes and if you need any support.

SunboX, 2022-03-19:
@gioblu Thanks for your reply! Both were ESP32, not ESP8266. But thanks, I will try this and let you know the result!

gioblu, 2022-03-19:
Sorry @SunboX in this case you should modify the constants here: https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/Timing.h#L207

SunboX, 2022-03-24:
Great! Seems I have success using this configuration on both ESP32 using pin 25:

```c++
/* Heltech WiFi LoRa ESP32, generic ESP32 --------------------------------- */
#if defined(ESP32)
  #if SWBB_MODE == 1
      /* Added full support to MODE 1 - 28/06/2018
         Working on pin: 12 and 25 */
      #define SWBB_BIT_WIDTH   46
      #define SWBB_BIT_SPACER 112
      #define SWBB_ACCEPTANCE  40
      #define SWBB_READ_DELAY -10
    #endif
#endif
```

gioblu, 2022-03-30:
@SunboX I don't have a way to test it now, but your test result shows it is working perfectly.
I am not sure how I did not spot this in the last test round.

gioblu, 2022-04-03:
@SunboX can I ask you to execute a test that would help me to verify your config?
I would need to know if your config works also when ESP32 transmits to or receives from Arduino Nano/Uno or Mega.
This would verify that the configuration found works properly in both cases (ESP to ESP and ESP to arduino)

SunboX, 2022-04-03:
Hey, yes. That's no problem. I should have some Arduino in one of my boxes. I can have a look in some hours which ones I have and can tell you.

SunboX, 2022-04-03:
Ok, I have those to work with:

[image]

[image]

[image]

[image]

[image]

[image]

Please tell me which one I should use and please send me the test code (GitHub Gist or so). I will send you the results.

gioblu, 2022-04-03:
The Arduino duemilanove is OK.
You can use the networkAnalysis sketch: 
https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/NetworkAnalysis

SunboX, 2022-04-06:
@gioblu Yesterday I tested the UNO (for some reason I could not flash my Duemilanove with macOS Monterey) together with the ESP32. It did not work. ;) I will investigate when I have a bit more time.

2114L3, 2023-09-18:
try to remove the delay(1) from the main loop, put the send() to a counter to trigger it, and move all the serial output onto an event or counter to trigger it.

you have alot happening in the main loop that might impact update() from running freely as bitbang requires.

you are using send() very frequently which is non-blocking and has its own repeats, you might be maxing out a buffer or the bandwidth available on the wire.