gioblu/PJON #406: Undefined behavior when using Ports

rainerschoe, 2022-01-11:
Hi,
I am using PJON in a network, where a Device sometimes sends packets with a Port (value `2` or `1`) and sometimes without (calling `send()` with defaults which should be `PJON_BROADCAST`)

On the Receiving side (via [PjonHL](https://github.com/rainerschoe/PjonHL)), I then check the port if it is `2` or `1` and do special operations for those. All other packets go into default/legacy/compatibility processsing mode.

I noticed that everything works fine, until The first message arrives on the receiver with port `1` or `2`. After this subsequent messages with `PJON_BROADCAST` AKA no port will be seen as if having port `1` or `2`.

I did some digging into PJON code and found the following:
parse_header only sets the port field in `PJON_Packet_Info` when `PJON_PORT_BIT` is set:
```
  static void parse_header(const uint8_t *packet, PJON_Packet_Info &info) {
.....
    #if(PJON_INCLUDE_PORT)
      if(info.header & PJON_PORT_BIT) {
        info.port = (packet[index] << 8) | (packet[index + 1] & 0xFF);
        index += 2;
      }
    #endif
```
However, when using `PJON_BROADCAST` AKA no port, this bit is not set in the header:
```
  static uint16_t compose_packet(
    PJON_Packet_Info info,
    uint8_t *destination,
    const void *source,
    uint16_t length
  ) {
    uint8_t index = 0;
    if(length > 255) info.header |= PJON_EXT_LEN_BIT;
    #if(PJON_INCLUDE_PORT)
      if(info.port != PJON_BROADCAST) info.header |= PJON_PORT_BIT;
      if((info.header & PJON_PORT_BIT) && (info.port == PJON_BROADCAST))
        info.header &= ~PJON_PORT_BIT;
    #endif
    ...
```
as the `PJON_Packet_Info` instance is re-used for all received packages:
```
PJON_Packet_Info last_packet_info;
```
The port field is still set to the value it had when the last package was received.

In my case this would be either `1`, `2`, or `PJON_BROADCAST` depending on which packet was sent before. Where if it gets `1` or `2` it will never be able to go back to `PJON_BROADCAST`

gioblu, 2022-01-15:
Ciao @rainerschoe happy new year and thank you very much for your report. Yes this is indeed me doing a stupid mistake. I will dig into the code and report back.

gioblu, 2022-01-31:
Ciao @rainerschoe you are right, thank you very much for your detailed report. You are right, the solution you propose is OK, so the old port value will be deleted as expected.

If you are willing to create a PR to fix this that would be a great help, I am fully taken by work in this period, so I have very few moments to focus on PJON.

rainerschoe, 2022-01-31:
Hi @gioblu thanks. Will create a PR, might take some days though, as I also have quite some things to do :package:P

FYI: I recently stabilized [PjonHL](https://github.com/rainerschoe/PjonHL) a little bit and now use it successfully in production environment (my smart-home ^^). However it is still not feature-complete and some rough around the edges. Unfortunately I have very little time and too much Projects running in parallel here :-), so progress is quite slow.

gioblu, 2022-01-31:
Ciao @rainerschoe thank you very much. I am very happy PJON is serving you well. 
I am sorry again for this stupid bug :) 

I think would be nice to record your contribution through a PR, although if you do not have the will or the time I can do that ASAP, at the end the change requires to add just one line of code.

I can add PjonHL to the wiki when you consider it is ready :)
Compliments again for the job done on PjonHL

rainerschoe, 2022-01-31:
Will do :) I plan to testing the changes before creating a PR :P
If you could add PjonHL to the wiki (basic protocol is supported and tested) It would be nice, as it might generate some traffic and contributions.

PR #408 (2022-02-10): Fix undefined behavior when using ports #406
fixes #406 
I tested this in my setup, using port 0,1 and 2. Now receiving works again :)
I also implemented it a little bit different that proposed in the issue.
I now use explicit re-initialization of the PacketInfo variable to avoid similar problems in other scenarios (packet-id, bus-id) which in my opinition might suffer from the same problem.
closed 2022-02-10 by gioblu