gioblu/PJON #409: Question: How to read a character array from a received struct.

VeldtDataConsultancy, 2022-02-18:
Good day.

I'm using PJON for a short while and I already love it a lot. The documentation is very good and I love the single wire transmission. I'm currently trying PJON in a little bit more professional fashion, but I'm running into a small issue when I want to read a received struct with a character array.

I’m trying to setup a simple PJON network. In this network I’m using an ESP-32 as a command module, and several Arduino nano’s as worker modules that control single parts of an escape box I’m building. The Arduinos only have to send their state to the ESP-32, and the ESP-32 is telling the nano’s what to do. To make this possible I make use of the PJON SerialBitBang library. The PJON documentation is really good and I love the single wire communication.

To make the communication possible, I want to make use of a simple struct consisting of an integer and a small character array.

```
struct payLoad {
  uint8_t cmd;
  char msgLine[20];
};
```

When I let the receiver print the struct, the integer gets printed fine. However, when I want to print the msgLine char array, I see the following:
Serial Monitor output:

```
10
1{
```
 
The integer (10) shows as sent.
The character array “12345” however is not. Only the 1 is shown.

*So my (maybe beginner) question: Could you or someone please point out to me what I am doing wrong here?* 
I'm near 100% sure that this is not a PJON related issue, but rather me missing a detail here.
I looked in the PJON documentation and there is an example for integers within a struct, but I could not find an example for a character array.

Code that can be used to reproduce the issue:
Arduino worker:

```
#include <PJONSoftwareBitBang.h>
PJONSoftwareBitBang bus(19);

struct payLoad {
  uint8_t cmd;
  char msgLine[20];
};

payLoad pl;

void send_command(uint8_t id, uint8_t cmd, char msgLine[20]) {
  payLoad pl;
  strcpy(pl.msgLine , msgLine);
  pl.cmd = cmd;
  bus.send(id, &pl, sizeof(&pl));
};

void setup() {
  bus.strategy.set_pin(8);
  bus.begin();
}

void loop() {
  send_command(20,10,"12345");
  bus.update();
  delay(2000);
}
```

Code for the receiving ESP-32:

```
// Command module running on an ESP32.
#include <PJONSoftwareBitBang.h>
PJONSoftwareBitBang bus(20);

// STRUCTS
struct payLoad {
  uint8_t cmd;
  char msgLine[20];
};

payLoad pl;

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  memcpy(&pl, payload, sizeof(pl));
  Serial.println(pl.cmd);
  Serial.println(pl.msgLine);
};

void setup() {
  Serial.begin(9600);
  bus.strategy.set_pin(25);
  bus.set_receiver(receiver_function);
  bus.begin();
}

void loop() {
  bus.receive(10000);
}
```

fredilarsen, 2022-02-18:
Hi, I guess this could have to do with padding/address alignment, depending on the architecture. If on a 32 bit architecture, the uint8_t will be padded with 3 bytes so that next member starts on a 32 bit aligned address. So could get some garbage between the members, and this varies between 64 bit , 32 bit and 16 bit architectures.
Sending a structure is generally fine as long as the number of bits in the architecture on each side is the same.

Try to copy everything to a tightly packed byte array and send that.
uint8_t packet[21];
memcpy(packet, msg.cmd, 1);
memcpy(&packet[1], msg.msgLine, 20);

Then you have a tightly packet array of length 21, that can be decoded in the same way on the other side.

VeldtDataConsultancy, 2022-02-18:
Awesome @fredilarsen.

Thanks a lot with your quick answer. 
Let me try it out. Once I got a result I will come back.

VeldtDataConsultancy, 2022-02-19:
@fredilarsen. Man, now that I know the answer for this issue I'm ashamed to mention it to you. I found out what the issue is while I worked with your feedback. Classic development example of how one character can ruin your day (e.g. read the **** manual). At the moment of sending I used the size of the pointer as payload size, instead of the actual struct.

So in the sender code I had:

`bus.send(id, &pl, sizeof(&pl));`

But it should have been:

`bus.send(id, &pl, sizeof(pl));`

By removing the & in the size turns the payload size to 21 byte instead of 2.

It works like a charm now. Also from Arduino to ESP-32 and the other way around.

Thanks a lot for your help and feedback. Once again love what you are doing here.

closed 2022-02-19 by VeldtDataConsultancy