gioblu/PJON #412: PJON on RP2040

fletchjp, 2022-04-06:
Is there any plan for PJON to be supported on the RP2040? I would like to work with Raspberry Pi Pico and Arduino NANO RP2040 connect. I have naively compiled for the Raspberry Pi Pico using PJON library on the Arduino IDE. It tells me RP2040 is not supported although it does complle the Serial code.

gioblu, 2022-04-10:
Ciao @fletchjp give it a try, it may work out of the box, I do not have the chance now to test it myself.

fletchjp, 2022-04-10:
Thank you for the suggestion. I have had a try at this with the Bitbang test with the error handler in the transmitter.
I get Connection lost with device id 44 and nothing in the receiver.
I have looked into the code and something is needed for ARDUINO_ARCH_RP2040. I don't know enough about how your code works to know what to put where. At the moment compiling generates this warning:
WARNING: library PJON claims to run on atmelavr, avr, esp8266, esp32, sam, samd, teensy, tiny, stm32, STM32F1 architecture(s) and may be incompatible with your current board which runs on rp2040 architecture(s).
I don't know where this comes from. Where should I look to support this case?
John

fletchjp, 2022-04-11:
I thing what is needed is some configuration information. Is there any information about what is needed? John