gioblu/PJON #416: PJONSoftwareBitBang between stm32f4 and arduino nano not working

rozrabiak, 2022-05-20:
Communication between two arduino nanos working.
That same code in stm32f4 (stm32duino) not working, arduino nano says: Connection with device ID 10 is lost.
I'm using WeAct BlackPill pin PC14 (5V tolerant) and Arduino Nano PD5.

**ARDUINO NANO CODE:**

#include <PJONSoftwareBitBang.h> // Include only SoftwareBitBang
#define pjonPin 5   //pin komunikacyjny pjon

PJONSoftwareBitBang pjonBus(43); 

void pjonReceiver(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info);
void error_handler(uint8_t code, uint16_t data, void *custom_pointer) {
  if(code == PJON_CONNECTION_LOST) {
    Serial.print("Connection with device ID ");
    Serial.print(pjonBus.packets[data].content[0], DEC);
    Serial.println(" is lost.");
  }
  if(code == PJON_PACKETS_BUFFER_FULL) {
    Serial.print("Packet buffer is full, has now a length of ");
    Serial.println(data, DEC);
    Serial.println("Possible wrong bus configuration!");
    Serial.println("higher PJON_MAX_PACKETS if necessary.");
  }
  if(code == PJON_CONTENT_TOO_LONG) {
    Serial.print("Content is too long, length: ");
    Serial.println(data);
  }
};
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
    pjonBus.set_error(error_handler);
    pjonBus.strategy.set_pin(pjonPin);
    pjonBus.set_receiver(pjonReceiver);
    pjonBus.begin();   
    Serial.println("START");
}

void loop() {
  // put your main code here, to run repeatedly:

    pjonBus.update();
    pjonBus.receive(1000);
pjonBus.send(10, "test", 4);
delay(1000);
}

void pjonReceiver(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info)
{
    Serial.print("dane ze sterownika ");
    Serial.println(packet_info.tx.id);
    Serial.print("dane ");
    Serial.println(payload[0]);  
};

**CODE FROM STM32DUINO:**

PJONSoftwareBitBang pjonBus(10); 
#define pjonPin PC14

void setup()
{
    pjonBus.strategy.set_pin(pjonPin);
    pjonBus.begin();
    pjonBus.set_receiver(pjonReceiver);
    pjonBus.set_error(pjon_error_handler);
}

void loop()
{
    pjonBus.update();
    pjonBus.receive(); //1000
}

void pjonReceiver(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info)
{
    Serial.print("dane z pjon ");
    Serial.println(payload[0]);
};

gryzli133, 2022-06-07:
Did you try to use other pin on stm32? I had similar problems with ESP8266, but then I have used another pin and it starts to work.

rozrabiak, 2022-10-10:
> Did you try to use other pin on stm32? I had similar problems with ESP8266, but then I have used another pin and it starts to work.

I'll try all pins :-)
F4 is on unsuporrted list with SBB...

2114L3, 2023-09-18:
dont use delay in your main loop, you will be holding the nano out from running update() as it needs to for the non-blocking send command. keep the main loop running and use counters to trigger your send