gioblu/PJON #420: Unable to compile MQTT examples

JeroenVanHoye, 2022-11-24:
When trying to compile examples that make use of PJONMQTTTranslate.h the Arduino IDE throws an error that it can not find the ReconnectingMqttClient.h library.

What needs to be installed for this to work?

Many thanks in advance!

fredilarsen, 2023-05-02:
You need this repo: https://github.com/fredilarsen/ReconnectingMqttClient
Download it and unzip it to your Arduino/libararies folder.

2114L3, 2023-05-14:
This doesnt seem to be working with the example > PJON > ESP8266 > Local > MQTTTranslate > EnvironmentalController. 
The ReconnectingMqttClient uses ethernetlink which isnt supported on ESP8266/ESP32 re the wiki

compiling it throws errors on ntohl, htonl, htons, etc 
is there something else missing? or is MQTTTranslate not possible on platforms unsupported by ET?

fredilarsen, 2023-05-14:
@2114L3 I can confirm that the ReconnectingMqttClient does work on ESP8266, I just used it in my l[atest project](https://github.com/fredilarsen/OrigoSmokeDetector/tree/main/examples/OrigoMqtt). You could have a look at the include files there in case you missed something.
I did use Arduino IDE 1.8.6, though, which is rather old.

The dependency to PJONEthernetTCP.h is to pull in some system dependent defines. It does not use PJONEthernetTCP socket communication code.

What environment are you using?

2114L3, 2023-05-15:
Thank you!  

I was:
Arduino IDE 1.8.19 (linux)
ReconnectingMqttClient 1.2.0
ESP8266 Community 2.4.1
"WeMos D1 R2 & mini"

noting your board quote has LOLIN and mine did not - I've spotted i was on ESP3266 community 2.4.1, updating this to 3.1.2 has resolved the issue and compiling. 

regarding the wiki listing ET unsupported for ESP*, as EthernetLink is from within ET had me worried and lead me wrong.

hope this helps the original question.

closed 2023-08-08 by gioblu