gioblu/PJON #431: Moving from V12 to V13 gives significant new compiler warnings `Wregister` with C++17

sticilface, 2024-07-29:
I've moved to V13 and I now get a large number of complier warnings (See below), when using C++17 on platformio. 
The errors do not occur with C++11, or in the Arduino IDE.  
nonetheless it would be good to fix if possible to keep compatibility moving forward.  
The arduinoIDE is still using C++11. 

```In file included from .pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_ARDUINO_Interface.h:35:0,
                 from .pio/libdeps/mega_C_RELAY/PJON/src/interfaces/PJON_Interfaces.h:4,
                 from .pio/libdeps/mega_C_RELAY/PJON/src/PJON.h:61,
                 from .pio/libdeps/mega_C_RELAY/PJON/src/PJONSoftwareBitBang.h:4,
                 from src/mega_C_RELAY/main.cpp:2:
.pio/libdeps/mega_C_RELAY/PJON/src/strategies/SoftwareBitBang/SoftwareBitBang.h: In member function 'bool SoftwareBitBang::can_start()':
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:277:24: warning: ISO C++1z does not allow 'register' storage class specifier [-Wregister]
       uint8_t register saveSreg = SREG; \
                        ^
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:297:9: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
         PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_DDR_REG(P), P, V) \
         ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/strategies/SoftwareBitBang/SoftwareBitBang.h:71:7: note: in expansion of macro 'PJON_IO_MODE'
       PJON_IO_MODE(_input_pin, INPUT);
       ^~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/strategies/SoftwareBitBang/SoftwareBitBang.h: In member function 'uint16_t SoftwareBitBang::receive_response()':
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:277:24: warning: ISO C++1z does not allow 'register' storage class specifier [-Wregister]
       uint8_t register saveSreg = SREG; \
                        ^
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:288:11: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
           PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_PORT_REG(P), P, V) \
           ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/strategies/SoftwareBitBang/SoftwareBitBang.h:151:9: note: in expansion of macro 'PJON_IO_WRITE'
         PJON_IO_WRITE(_output_pin, LOW);
         ^~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:277:24: warning: ISO C++1z does not allow 'register' storage class specifier [-Wregister]
       uint8_t register saveSreg = SREG; \
                        ^
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:288:11: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
           PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_PORT_REG(P), P, V) \
           ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/strategies/SoftwareBitBang/SoftwareBitBang.h:155:9: note: in expansion of macro 'PJON_IO_WRITE'
         PJON_IO_WRITE(_input_pin, LOW);
         ^~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:277:24: warning: ISO C++1z does not allow 'register' storage class specifier [-Wregister]
       uint8_t register saveSreg = SREG; \
                        ^
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:297:9: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
         PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_DDR_REG(P), P, V) \
         ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/strategies/SoftwareBitBang/SoftwareBitBang.h:158:11: note: in expansion of macro 'PJON_IO_MODE'
           PJON_IO_MODE(_output_pin, OUTPUT);
           ^~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:277:24: warning: ISO C++1z does not allow 'register' storage class specifier [-Wregister]
       uint8_t register saveSreg = SREG; \
                        ^
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:288:11: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
           PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_PORT_REG(P), P, V) \
           ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/strategies/SoftwareBitBang/SoftwareBitBang.h:159:11: note: in expansion of macro 'PJON_IO_WRITE'
           PJON_IO_WRITE(_output_pin, HIGH);
           ^~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:277:24: warning: ISO C++1z does not allow 'register' storage class specifier [-Wregister]
       uint8_t register saveSreg = SREG; \
                        ^
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:288:11: note: in expansion of macro 'PJON_IO_ATOMIC_WRITE'
           PJON_IO_ATOMIC_WRITE((uint8_t*) PJON_IO_PIN_TO_PORT_REG(P), P, V) \
           ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:312:9: note: in expansion of macro 'PJON_IO_WRITE'
         PJON_IO_WRITE(P, LOW); \
         ^~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/strategies/SoftwareBitBang/SoftwareBitBang.h:161:11: note: in expansion of macro 'PJON_IO_PULL_DOWN'
           PJON_IO_PULL_DOWN(_output_pin);
           ^~~~~~~~~~~~~~~~~
.pio/libdeps/mega_C_RELAY/PJON/src/interfaces/ARDUINO/PJON_IO.h:277:24: warning: ISO C++1z does not allow 'register' storage class specifier [-Wregister]
[... 274 lines cut ...]
Alternatives for PJONSoftwareBitBang.h: [PJON@13.1]
ResolveLibrary(PJONSoftwareBitBang.h)
  -> candidates: [PJON@13.1]
/Users/amelvin/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/amelvin/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/amelvin/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/amelvin/Documents/Arduino/libraries/PJON/src /private/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino/sketches/B0B6BC7686C0908AFE1F65B0201F3C86/sketch/Device1.ino.cpp -o /dev/null
Using cached library dependencies for file: /Users/amelvin/Documents/Arduino/libraries/PJON/src/interfaces/WINX86/Serial/Serial.cpp
Generating function prototypes...
/Users/amelvin/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/amelvin/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/amelvin/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/amelvin/Documents/Arduino/libraries/PJON/src /private/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino/sketches/B0B6BC7686C0908AFE1F65B0201F3C86/sketch/Device1.ino.cpp -o /private/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/484919799/sketch_merged.cpp
/Users/amelvin/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/484919799/sketch_merged.cpp
Compiling sketch...
/Users/amelvin/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/amelvin/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino -I/Users/amelvin/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/standard -I/Users/amelvin/Documents/Arduino/libraries/PJON/src /private/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino/sketches/B0B6BC7686C0908AFE1F65B0201F3C86/sketch/Device1.ino.cpp -o /private/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino/sketches/B0B6BC7686C0908AFE1F65B0201F3C86/sketch/Device1.ino.cpp.o
Compiling libraries...
Compiling library "PJON"
Using previously compiled file: /private/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino/sketches/B0B6BC7686C0908AFE1F65B0201F3C86/libraries/PJON/interfaces/WINX86/Serial/Serial.cpp.o
Compiling core...
[... cut ...]
Using library PJON at version 13.1 in folder: /Users/amelvin/Documents/Arduino/libraries/PJON 
/Users/amelvin/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size -A /private/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino/sketches/B0B6BC7686C0908AFE1F65B0201F3C86/Device1.ino.elf
Sketch uses 7034 bytes (21%) of program storage space. Maximum is 32256 bytes.
Global variables use 776 bytes (37%) of dynamic memory, leaving 1272 bytes for local variables. Maximum is 2048 bytes.
```

sticilface, 2024-09-06:
That does appear to fix the problem.  Thank you.  Should have found that myself... really...

fredilarsen, 2024-09-06:
The register keyword is a hint to the compiler, advising it to store that variable in a processor register instead of memory. It could speed up the generated code, but it seems now to be deprecated. I am not sure if modern processor architectures are the reason for this, or if the compiler now determines when to use processor registers automatically.