gioblu/PJON #435: Anyone uses PJON SBB with PCINT?

willie68, 2025-02-11:
Hi,

I'm working on a small game that uses up to 16 different MCUs for the parts of the game that need to communicate with each other. (It's called Hi-Tane, a hardware version of the popular Ktane game.)

Each module runs its own game part and gives the results to a central controller that manages the entire game state. Some modules need the current game state (e.g. the timestamp or the number of failed attempts) which the controller sends every second in a heartbeat message.

It's important that the communication is as error-free as possible. Unfortunately, with the standard SBB implementation and the actual implementation of the modules and controller, I have almost 80% to 90% communication loss. This is mainly due to the method for receiving not being called frequently enough.

In my current module sources, there are almost no other interrupt-driven parts. That's why it would be an advantage for my project if the PJON-SBB receiver part could be triggered via a pin change interrupt.

My question: Has anyone already done a project with similar requirements or a similar test with it?

I could still switch the communication to RS485, so the question here is whether it is worth continuing to use PJON for this project or not.

fredilarsen, 2025-02-11:
I did have one time-consuming function in one of my sketches that use PJON SWBB, and did call receive before and after that function but ended up with bad communication because of the long non-listening intervals.
I ended up inserting receive calls in multiple places in my time-consuming function, to reduce the non-listening gaps.
If this is not possible for you, perhaps you should use some hardware-buffered communication instead, like the PJON TCP or UDP strategies, or as you say, RS485.

I am not sure if interrupt driven PJON is possible. I suspect calling receive() within an interrupt may cause problems. A variant of PJON that is based on interrupts for some of the strategies can probably be created, if anybody are interested in doing this.

gioblu, 2025-02-11:
Ciao @willie68 I am happy to hear from you. I have never implemented an interrupt driven implementation of PJDL, although that would have been a very useful feature for many use cases. What i have learned looking at the project of PJON users is that you very rarely effectively need interrupts. 

Depending how you design your distributed system you may or may not need interrupts for it to operate nominally.

Obviously the best solution is to reduce the duration of the loop, one obvious example, if you rely on analogRead you can reduce its duration of orders of magnitude just configuring the prescaler, see https://forum.arduino.cc/t/faster-analog-read/6604/3  

Please try to:

1. Run networkAnalysis tests between devices to see if there is any interference or hardware connection problem
2. Run networkAnalysis tx towards your slaves running their definitive code

If you spot low quality while running test 2, try to:

1. Increase the amount of time the receive function polls, instead of `receive()` try `receive(1000)` or `receive(10000)`
2. Set a higher `SWBB_PREAMBLE`, that should help reducing packet loss while also reducing the maximum available bandwidth.

Another solution may be to change how the nodes communicate, instead of trying to catch with slaves a repeated state broadcast, let the master send a packet directed to each node cyclically containing the state and maybe let the slave answer with useful information if required. Doing so you can let the master be ready to receive the response and avoid packet loss when slaves communicate to the master.

willie68, 2025-02-11:
Thanks for your answers,
the problem is, every game module has a very different behavier.
1 module uses A/D converter, the other an 8x8 RGB Matrix. The third uses an OLED Display, showing texts. 
I have tried to add multiple receive, normally with duration 100. It's better, but for some modules not good enough. 
For me at this moment, it's more a question of 
“Is PJON basically a good choice for my problem or not?” I'm not at the point to optimize this yet.

As you know i have already other projects based on PJON with very good results. My foot controller and the model railway decoder (this is one master with 12 slave nodes) working great.
A major difference between them and the game is the load on the nodes. With the foot controller or the model railway decoder, there are normally no other long processes running on the node. Both basically just wait for the next command via PJON and then execute it. More like actors or sensors.
With the game module, the actual game is of course running on the node on first and PJON only has a short period of time to react. That's why I'm asking for an interrupt-controlled variant.

closed 2025-02-15 by willie68

fredilarsen, 2025-02-15:
> I've been playing around with the interrupts and PJON for a few days now.
> Unfortunately without success.
>
> With the INT0, whether on rising or change, PJON cannot receive anything.
> Something is preventing the receive from synchronizing with the signal.
> With a PCINT, it works a little better, but unfortunately I still get an
> error rate of around 80%. (For the heartbeat, I can't use anything other
> than a broadcast message, because this is very time-critical. I've had it
> without broadcast before, but then the sequence was no longer guaranteed
> and the timing was no longer correct at all.)
> Without an interrupt with an increase in the polling rate, the broadcasts
> are still only received every now and then. It seems to be more luck if the
> receiver is on receive and then a broadcast message comes. A receive(1000)
> is not possible in either the controller or the modules. They have other
> things to do at the same time. Updating the LC display, RGB LEDs and
> managing the actual games. Unfortunately, I can't get anything out of it.
> The next thing I did was play around with the RS485 modules and PJON. That
> only works well if I use the hardware serial. Unfortunately, I need it for
> other things, so I can't use it with some game modules. And collision
> detection was another blocking part. I also have to be careful when
> uploading that I turn off modules that aren't being used. I've already
> bricked a Nano.
> Conclusion: In this application, a hardware-based communication solution
> makes more sense. So I'll probably build the whole thing with CAN-BUS
> modules. Seems to make the most sense for this application.

What about Ethernet, having hardware buffers?

willie68, 2025-02-15:
Hmmm, i think Ethernet is not an option. All game modules are of size 8x8x4cm. 6 modules plus the controller have to fit into a small case the size of a DIN A4 sheet of paper (https://www.pollin.de/p/kunststoff-geraetekoffer-330x280x125-mm-schwarz-501265) 
If I add the patch cables plus the 8-way switch, the whole thing won't fit in the case.